autowarefoundation/autoware_universe #3229: Lane change module development plan

rej55, 2023-03-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Lane change module has features to plan lane changing to arrive goal.
The features implemented in the current lane change module are following:

- Driving lane planning
  - To arrive goal, it plans when and where to change lanes
  - The module also has a feature to change lanes by a request from operators
- Safety check
- Blinker control
- Path generation (now improving)
- Aborting lane change (now improving)

To make it more efficient and safer, some issues remain:

- Improving path planning
  - Path planning more like a real driver to improve feeling of riding
- Improving aborting lane change
  - To abort more safely, improving aborting behavior planning is required
- Dynamic route planning
  - For example, detour if lane change is not possible due to traffic jams
  - Common issue with mission planner
- Interactions with other cars
  - For example, get in a car platoon in a traffic jam
  - Indication of intention by blinker

### Purpose

For more efficient and safer lane changes, the remained issues should be resolved.

### Possible approaches

We are currently working on the following issues:

- Improving path planning
  - Planning including acceleration during lane change
  - Consideration of curvature according to vehicle speed
- Improving aborting lane change
  - Scenario enumeration
  - Reconsider criteria for returning current lane or entering target lane

### Definition of done

Resolve remained issues.

TakaHoribe, 2023-07-06:
Recent updates.

Path generation
- https://github.com/autowarefoundation/autoware.universe/pull/3756
- https://github.com/autowarefoundation/autoware.universe/pull/4033
- https://github.com/autowarefoundation/autoware.universe/pull/3991
- https://github.com/autowarefoundation/autoware.universe/pull/3965
- https://github.com/autowarefoundation/autoware.universe/pull/3924
- https://github.com/autowarefoundation/autoware.universe/pull/3762
- https://github.com/autowarefoundation/autoware.universe/pull/3750
- https://github.com/autowarefoundation/autoware.universe/pull/3705

Safety check
- https://github.com/autowarefoundation/autoware.universe/pull/4105
- https://github.com/autowarefoundation/autoware.universe/pull/4172
- https://github.com/autowarefoundation/autoware.universe/pull/2704
- https://github.com/autowarefoundation/autoware.universe/pull/2704

Lane change cancel framework
- https://github.com/autowarefoundation/autoware.universe/pull/3566
- https://github.com/autowarefoundation/autoware.universe/pull/4080
- https://github.com/autowarefoundation/autoware.universe/pull/4083

Turn signal
- https://github.com/autowarefoundation/autoware.universe/pull/4115
- https://github.com/autowarefoundation/autoware.universe/pull/3983

mitsudome-r, 2023-08-30:
@TakaHoribe @shmpwk 
All the linked PRs seemed to be merged. Is this issue ready to be closed?

TakaHoribe, 2023-08-30:
@mitsudome-r Thank you for your comment. Yes, recently all lane change task has been completed. It would be nice if we could show the result of this development. 

@rej55 Can we close this issue? Is ok, please summarize the current status (what we can/cannot do) in this issue.

rej55, 2024-01-18:
# Lane change development results

## Completed items

### Improving path planning

- Planning including acceleration during lane change
- Consideration of curvature according to vehicle speed

### Improving aborting lane change

- Scenario enumeration
- Reconsider criteria for returning current lane or entering target lane

### Planning with obstacle avoidance

- Cooperating with avoidance module in behavior path planner

### Improving safety check

- Filtering objects to exclude unrelated objects

## Not completed items

### Blinker control for driver to driver interaction

### Aborting path generation

- Asymmetric path generation
  - Path generation without `PathShifter`

### Some improvements

- Path planning at terminal of lane
- Stabilize safety check functions

### Refactoring

closed 2024-01-18 by rej55