autowarefoundation/autoware_universe #3239: Adopt New ROS Node Configuration Lidar Apollo Segmentation TVM Nodes

kasperornmeck, 2023-04-01:
Make changes to [Lidar Apollo Segmentation TVM Nodes](https://github.com/autowarefoundation/autoware.universe/tree/main/perception/lidar_apollo_segmentation_tvm_nodes) in accordance to [Final3 Proposal](https://github.com/orgs/autowarefoundation/discussions/3371).

closed 2023-04-25 by angry-crab

Croquembouche, 2023-04-26:
Quick question about this package. Is this package only used if the Apollo LiDARs are used? What happens if I don't build this package?

angry-crab, 2023-04-27:
> Quick question about this package. Is this package only used if the Apollo LiDARs are used? What happens if I don't build this package?

It has nothing to do Apollo Lidars. This package is used for object detection. You can use Centerpoint instead.

PR #3390 (2023-04-25): refactor(lidar_apollo_segmentation_tvm): rework parameters
## Description

Implement the ROS Node configuration layout described in https://github.com/orgs/autowarefoundation/discussions/3371 for the `lidar_apollo_segmentation_tvm_nodes` package.

- Remove duplication of parameter declaration between the source code and the parameter file.
- Remove the default value from the source code in order to ensure all parameter values are passed from the parameter files.
- Use types that match the ones returned from declare_parameter() in order to prevent confusion and potential coding errors.
- Move t [... cut]