autowarefoundation/autoware_universe #3244: Raise error when `pointcloud_preprocessor` filters do not support `indices`

VRichardJP, 2023-04-03:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Several filters within `pointcloud_preprocessor` do not support `indices`, yet they don't raise any error if indices are provided.

For example:

https://github.com/autowarefoundation/autoware.universe/blob/70cf872517cad0e92cdbd92510c8ac28333efeb6/sensing/pointcloud_preprocessor/src/crop_box_filter/crop_box_filter_nodelet.cpp#L114-L116

Although indices are not really used in Autoware, it would be nice to raise an error when `indices` are provided but the implementation does not support it. Something like an exception would be acceptable I think.

### Expected behavior

Better crash than silently produce garbage data.

### Actual behavior

Ignores `indices`

### Steps to reproduce

N/A

VRichardJP, 2023-04-04:
Actually, there is only one filter that support indices `PassThroughFilterUInt16Component`, and it does not look like it's used anywhere in the code base.
```bash
$ rg IndicesPtr
filter.cpp
177:  const PointCloud2ConstPtr & input, const IndicesPtr & indices)
282:  IndicesPtr vindices;
441:  IndicesPtr vindices;
461:  const PointCloud2ConstPtr & input, const IndicesPtr & indices, PointCloud2 & output,

downsample_filter/random_downsample_filter_nodelet.cpp
70:  const PointCloud2ConstPtr & input, const IndicesPtr & /*indices*/, PointCloud2 & output)

downsample_filter/approximate_downsample_filter_nodelet.cpp
78:  const PointCloud2ConstPtr & input, const IndicesPtr & /*indices*/, PointCloud2 & output)

downsample_filter/voxel_grid_downsample_filter_nodelet.cpp
78:  const PointCloud2ConstPtr & input, const IndicesPtr & /*indices*/, PointCloud2 & output)

outlier_filter/ring_outlier_filter_nodelet.cpp
48:  const PointCloud2ConstPtr & input, [[maybe_unused]] const IndicesPtr & indices,

outlier_filter/radius_search_2d_outlier_filter_nodelet.cpp
43:  const PointCloud2ConstPtr & input, [[maybe_unused]] const IndicesPtr & indices,

crop_box_filter/crop_box_filter_nodelet.cpp
105:  const PointCloud2ConstPtr & input, const IndicesPtr & indices, PointCloud2 & output)
115:  const PointCloud2ConstPtr & input, [[maybe_unused]] const IndicesPtr & indices,

outlier_filter/voxel_grid_outlier_filter_nodelet.cpp
42:  const PointCloud2ConstPtr & input, [[maybe_unused]] const IndicesPtr & indices,

outlier_filter/dual_return_outlier_filter_nodelet.cpp
115:  const PointCloud2ConstPtr & input, [[maybe_unused]] const IndicesPtr & indices,

passthrough_filter/passthrough_filter_nodelet.cpp
70:  const PointCloud2ConstPtr & input, [[maybe_unused]] const IndicesPtr & indices,

blockage_diag/blockage_diag_nodelet.cpp
109:  const PointCloud2ConstPtr & input, [[maybe_unused]] const IndicesPtr & indices,

passthrough_filter/passthrough_filter_uint16_nodelet.cpp
49:  const PointCloud2ConstPtr & input, const IndicesPtr & indices, PointCloud2 & output)

pointcloud_accumulator/pointcloud_accumulator_nodelet.cpp
37:  const PointCloud2ConstPtr & input, [[maybe_unused]] const IndicesPtr & indices,

vector_map_filter/vector_map_inside_area_filter.cpp
80:  const PointCloud2ConstPtr & input, [[maybe_unused]] const IndicesPtr & indices,

polygon_remover/polygon_remover.cpp
52:  const PointCloud2ConstPtr & input, [[maybe_unused]] const IndicesPtr & indices,
```

Except for that filter, I propose we modify all other filters as such:
```c++
void VoxelGridDownsampleFilterComponent::filter(
  const PointCloud2ConstPtr & input, const IndicesPtr & indices, PointCloud2 & output)
{
  if (indices) {
    throw std::runtime_error("indices are not supported yet");
  }
  // ...
```

yukkysaito, 2023-04-07:
>Several filters within pointcloud_preprocessor do not support indices, yet they don't raise any error if indices are provided.

@VRichardJP 
Let me check a little more.
Does it need to generate an error when indices is provided?
Is there a problem if it is just not used?

VRichardJP, 2023-04-07:
If the user provides indices, they expect it will be used. So I think it is bad to silently ignore indices if they are provided but not used. At least if a warning is provided, users will know their use case is not supported.

closed 2023-06-05 by VRichardJP