autowarefoundation/autoware_universe #3269: `ring_outlier_filter` blows up my cache (and yours too)

VRichardJP, 2023-04-04:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Current `ring_outlier_filter` implementation is slow. I have not run a precise benchmark, but this filter runs significantly slower than any other filter.
During a recent test drive with our demo vehicle, I observed that when `ring_outlier_filter` is disabled, htop CPU usage is reduced by 20% (!)

I am pretty sure it is a known problem, and @sykwer is working hard on improving the current implementation.
His changes already improve the performance a lot, but I have the feeling his approach does not tackle the true cause of the bad performance, so I think it is worth opening a separate issue.

Let's first look at how the current algorithm is implemented. Roughly it looks like this:
1. Iterate over the input cloud and group point indices by ring
2. For each ring:
2.1. iterate over the ring points, and group points belonging to the same "walk"
3.1. when a walk is closed, copy indexed points to the output cloud if the walk is long enough.

There is nothing wrong with the algorithm in itself. It looks fairly efficient.
It has a big problem though: it is not adapted to the data it is supposed to process.

Let me explain.
By design, LIDAR sensors naturally collect data in "azimut-major order". So unless one of the previous preprocessing node changes that order, the input cloud data of `ring_outlier_filter` always looks like this:
```
p[0] = {ring = 0; azimut = 0}
p[1] = {ring = 1; azimut = 0}
p[2] = {ring = 2; azimut = 0}
p[3] = {ring = 3; azimut = 0}
```

For example here, I have printed the first 100 point indices of ring 0 (from one rosbag of mine):
```
234000 236352 240432 283680 375552 376608 377664 536064 562800 563904 590160 596352 597360 744000 785136 798624 874032 875088 876144 877200 878256 879312 880272 881280 882288 883392 884544 885648 887904 889056 890208 891216 892128 893136 894048 894912 895824 896784 897696 898656 899664 900624 901680 902688 903552 904416 905376 906288 907152 908064 909024 910080 910992 912048 913056 913920 914832 915696 916608 917568 918384 919200 920112 921024 922080 922992 923856 924912 926112 927216 928272 929184 930144 931344 932256 933408 934464 935280 936240 938304 940560 941616 942576 943824 945024 946176 947520 948624 949776 950928 952176 955632 957072 958464 959664 960864 961824 964320 965616 966960
```

As you can see, consecutive points within the ring are **very** far from each other in memory. In the previous example, the minimum distance between 2 ring points is 816 bytes! the average distance is 7403 bytes!

If we have a N layers LIDAR sensor, 2 consecutive ring points are separated by `N*sizeof(PointXYZIRADRT)` bytes. So with a 32-layers LIDAR sensor, 2 consecutive points are seperated by around 1KB. This is what I measure.

My poor x86 has a L1 data cache of 32KB, and a L2 cache of 256KB, so the node totally blows up the cache. No wonder my CPU usage increases by 20% when I use `ring_outlier_filter`!

I think there is no choice but to find a way to rewrite the implementation in a cache friendly manner. For example, instead of exploring ring points separately, it could be faster to just process all the rings at the same time (i.e. compute the 128 walks simultaneously).

### Expected behavior

fast!

### Actual behavior

slow!

### Steps to reproduce

Run it!

VRichardJP, 2023-04-05:
Just to confirm there is a cache problem, I tried to measure cache performance with `perf`:

On a first terminal, I start the filter:
```bash
$ ros2 run pointcloud_preprocessor ring_outlier_filter_node --ros-args --param use_sim_time:=true
```

On a second terminal, I play a rosbag (redirecting one of the cloud topic to `/input`).

In a last terminal, I run `perf`:
```
$ perf stat -e task-clock,cycles,instructions,cache-references,cache-misses -p 79694 sleep 30
```

Normally it is possible to feed the command directly to `perf`. In my case I have my ros2 workspace in a docker container, and the reported data seems incorrect when I use perf from the container. So instead I run the node in the container and the `perf` command outside. I don't think it should have any impact.

When the rosbag is not running (idling), perf reports 16% cache miss (I am not sure why):
```
 Performance counter stats for process id '79694':

             10.84 msec task-clock                #    0.000 CPUs utilized          
         9,050,522      cycles                    #    0.835 GHz                    
         3,036,918      instructions              #    0.34  insn per cycle         
         1,254,614      cache-references          #  115.698 M/sec                  
           204,268      cache-misses              #   16.281 % of all cache refs    

      30.018013843 seconds time elapsed
```

When I play the rosbag, perf reports 26% of cache miss:
```
 Performance counter stats for process id '79694':

            881.88 msec task-clock                #    0.029 CPUs utilized          
     1,100,977,814      cycles                    #    1.248 GHz                    
       987,476,865      instructions              #    0.90  insn per cycle         
       127,555,006      cache-references          #  144.640 M/sec                  
        33,928,052      cache-misses              #   26.599 % of all cache refs    

      30.002586833 seconds time elapsed
```

To compare, I did the same experiment with `voxel_grid_downsample_filter_node`. When idle, cache miss is also at 16%, when the node is running, cache miss rate does not increase much:
```
 Performance counter stats for process id '82621':

          1,567.57 msec task-clock                #    0.052 CPUs utilized          
     1,612,515,626      cycles                    #    1.029 GHz                    
     1,667,993,816      instructions              #    1.03  insn per cycle         
       118,662,204      cache-references          #   75.698 M/sec                  
        22,285,384      cache-misses              #   18.781 % of all cache refs    

      30.003842033 seconds time elapsed
```

I also checked with `perf stat -a` to watch the entire system. When the `ring_outlier_filter_node` runs, the cache miss rate jumps by +10%.

VRichardJP, 2023-04-07:
One issue with the setup of my previous comment is that I run the filter alone. In practice, the filter runs within the `pointcloud_container`, in which filters share inputs/outputs thanks to ROS2 intra processing capabilities. In my previous experimentation, a good share of the cache misses were actually coming from the input/output data copy (which does not happen inside the container).

I think more fair to measure the filter performance when it is running inside the container. Here is a new setup:

- Terminal 1: rosbag play
```
$ ros2 bag play <rosbag> --topics /sensing/lidar/top/velodyne_packets
```

- Terminal 2: running the sample lidar pipeline
```
$ ros2 launch sensing_lidar.launch.xml
```

With the following launch file:
```xml
<launch>
  <include file="$(find-pkg-share autoware_launch)/launch/pointcloud_container.launch.py">
    <arg name="use_multithread" value="true"/>
    <arg name="container_name" value="pointcloud_container"/>
  </include>

  <include file="$(find-pkg-share global_parameter_loader)/launch/global_params.launch.py">
    <arg name="use_sim_time" value="true"/>
    <arg name="vehicle_model" value="sample_vehicle"/>
  </include>

  <group>
    <push-ros-namespace namespace="sensing"/>
    <include file="$(find-pkg-share sample_sensor_kit_launch)/launch/lidar.launch.xml">
      <arg name="launch_driver" value="false"/>
      <arg name="vehicle_mirror_param_file" value="$(find-pkg-share sample_vehicle_description)/config/mirror.param.yaml"/>
      <arg name="use_pointcloud_container" value="true"/>
      <arg name="pointcloud_container_name" value="pointcloud_container"/>
    </include>
  </group>

  <group>
    <include file="$(find-pkg-share tier4_vehicle_launch)/launch/vehicle.launch.xml">
      <arg name="vehicle_model" value="sample_vehicle"/>
      <arg name="sensor_model" value="sample_sensor_kit"/>
      <arg name="vehicle_id" value="default"/>
      <arg name="launch_vehicle_interface" value="false"/>
      <arg name="config_dir" value="$(find-pkg-share individual_params)/config/default/sample_sensor_kit"/>
    </include>
  </group>
</launch>
```

- Terminal 3: checking the filter performance
```
$ ros2 topic echo /sensing/lidar/top/ring_outlier_filter/debug/processing_time_ms --field data
```

- Terminal 4: recording with perf
```
$ perf record -e task-clock,cycles,instructions,cache-references,cache-misses -p `ps aux | grep pointcloud_container | head -n 1 | awk '{ print $2 }'` -- sleep 30
```

Now I get interesting results with `perf report`:

- Within the pipeline, `RingOutlierFilterComponent` is the filter with the highest cache miss count:
```
Samples: 10K of event 'cache-misses', Event count (approx.): 196286799
Overhead  Command          Shared Object                                            Symbol
  24.84%  component_conta  libc.so.6                                                [.] __memset_avx2_unaligned_erms
  12.59%  component_conta  libpointcloud_preprocessor_filter.so                     [.] pointcloud_preprocessor::RingOutlierFilterComponent::filter
   8.20%  component_conta  libpointcloud_preprocessor_filter.so                     [.] std::__detail::_Map_base<unsigned short, std::pair<unsigned short const, std::vector<unsigned long, std::allocator<unsigned long> >
   8.18%  component_conta  libc.so.6                                                [.] __memmove_avx_unaligned_erms
   5.36%  component_conta  libpointcloud_preprocessor_filter.so                     [.] pointcloud_preprocessor::CropBoxFilterComponent::faster_filter
```

As I expect further, I can see the biggest source of cache miss within the ring outlier filter comes from this line:
```
       │     float azimuth_diff = next_pt_azimuth - current_pt_azimuth;                                                                                                                                                   ▒
       │       movss    (%rdi,%rdx,1),%xmm1                                                                                                                                                                               ▒
 16.68 │       subss    (%rsi,%rcx,1),%xmm1                                                                                                                                                                               ▒
```
As I showed in my original report, 2 consecutive ring points are actually very far in memory, causing many cache miss.
The second biggest source of cache miss is from a `vector.push_back`:
```
       │     std::vector<unsigned long, std::allocator<unsigned long> >::push_back(unsigned long const&):                                                                                                                 ▒
       │     ++this->_M_impl._M_finish;                                                                                                                                                                                   ▒
 16.29 │       add      $0x8,%rsi                                                                                                                                                                                         ▒
       │       mov      %rsi,0x8(%rax)                                                                                                                                                                                    ▒
       │     pointcloud_preprocessor::RingOutlierFilterComponent::filter(std::shared_ptr<sensor_msgs::msg::PointCloud2_<std::allocator<void> > const> const&, std::shared_ptr<std::vector<int, std::allocator<int> > > con▒
```
There are several `push_back` in the code, but I guess most cache miss must come from this one (same issue as before):

https://github.com/autowarefoundation/autoware.universe/blob/a5c90a9e25b90a160e41daf595fbd21bd8a9c0f0/sensing/pointcloud_preprocessor/src/outlier_filter/ring_outlier_filter_nodelet.cpp#L105-L108

xmfcx, 2023-05-31:
@VRichardJP do you need any help on this topic?

What is the last situation on the PRs related in general? We can try to speed things up if they are waiting for review.

VRichardJP, 2023-06-01:
@xmfcx Hi

For a while I have waited for https://github.com/autowarefoundation/autoware.universe/pull/2618 and https://github.com/autowarefoundation/autoware.universe/pull/3014 to be merged (since this PR is built on top of both).

Recently I have rebased my work here: https://github.com/autowarefoundation/autoware.universe/commit/9a24f2ca035d634c97d0cfa161b1f8a3f36228da
But I did not have the time to re-benchmark it.

Also, it does not include new changes made by https://github.com/autowarefoundation/autoware.universe/pull/2618 earlier this week. In particular, @miursh changed its approach and removed the padding initially introduced to PointXYZIRADRT type (which I relied on).

It does not change much, but it means I need to:
- rebase https://github.com/autowarefoundation/autoware.universe/pull/3293 with latest https://github.com/autowarefoundation/autoware.universe/pull/2618 (i.e. change a few memcpy)
- test
- benchmark against current implementation

I don't have much time recently to work on Autoware, maybe I can find myself some free time to finish this next week.

VRichardJP, 2023-06-05:
@xmfcx I think it is ready for review

closed 2023-08-24 by VRichardJP