autowarefoundation/autoware_universe #3309: `pure_pursuit` controller node exit code -6

TauTheLepton, 2023-04-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When starting autoware with `lateral_controller_mode` set to `mpc` (default) everything works fine and in the end process finishes cleanly, but when i use `pure_pursuit` controller node process dies with exit code -6

### Expected behavior

Controller node finishes cleanly in `pure_pursuit` mode the same as in the `mpc` mode.

### Actual behavior

Process dies with exit code -6 (log only from this component):

```log
[component_container-33] [INFO] [1680858929.092014981] [rclcpp]: signal_handler(signum=2)
[component_container-33] corrupted double-linked list
[ERROR] [component_container-33]: process has died [pid 243809, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=control_container -r __ns:=/control -p use_sim_time:=False -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7'].
```

### Steps to reproduce

1. Build autoware with `--cmake-args -DCMAKE_BUILD_TYPE=Release` option.
2. Download map from [here](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/)
3. Launch autoware with specified controller mode:

```bash
ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit lateral_controller_mode:=pure_pursuit
```

### Versions

- OS: Ubuntu 22.04
- ROS: Humble
- Autoware: up to date while writing this issue (`autoware.universe` commit df450d187f7e7fba8ff055299a3ec3b202647b94)

### Additional context

I found out about this while working on an issue with smoke test #2684. You can see some [errors](https://github.com/autowarefoundation/autoware.universe/issues/2684#issuecomment-1495832805) I was getting there, especially the last one (pure pursuit, with delay).

takayuki5168, 2023-04-07:
In my environment, there is no problem on planning simulator with `lateral_controller_mode:=pure_pursuit`

TauTheLepton, 2023-04-07:
> In my environment, there is no problem on planning simulator with `lateral_controller_mode:=pure_pursuit`

I'm not sure if I expressed myself clearly enough, this problem occurs after I close autoware. Only then the process dies with exit code -6 instead of finishing cleanly. 

I did more experiment and it seems that I get this error every time if I close autoware after actually doing something with it. If I open autoware and immediately close it the error appears only sometimes, depending on timing.

takayuki5168, 2023-04-07:
> after I close autoware

Sorry, I missed it. If you wanna fix it, you can with `gdb`.

TauTheLepton, 2023-09-04:
I have closed the issue as the fix PR has been merged.

closed 2023-09-04 by TauTheLepton