autowarefoundation/autoware_universe #3318: clearly state and check assumptions on input data of `pointcloud_preprocessor` filters

VRichardJP, 2023-04-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

As a way to improve filters reliability, and after https://github.com/autowarefoundation/autoware.universe/pull/3306 is merged, I would like to add proper assumption checks in `pointcloud_preprocessor` filters.

As explained in https://github.com/autowarefoundation/autoware.universe/pull/3306 and https://github.com/autowarefoundation/autoware.universe/pull/2618, several `pointcloud_preprocessor` filters make implicit assumptions on the input data. In most cases these assumptions are correct but sometimes it is not the case. And when it is not, it is actually really difficult to find it out.

For example `ring_outlier_filter` intensity output [has been wrong for at least a year](https://github.com/autowarefoundation/autoware.universe/pull/780) without anyone noticing.

### Purpose

Improve filters reliability.

### Possible approaches

A simple solution is to add a simple check in each `filter` function that makes implicit assumptions on the input data, like this:
```C++
void RingOutlierFilterComponent::filter(
  const PointCloud2ConstPtr & input, [[maybe_unused]] const IndicesPtr & indices,
  PointCloud2 & output)
{
  // make sure we can memcpy from input->data to PointXYZI type
  if (!utils::is_data_layout_compatible_with_PointXYZI(*input)) {
    RCLCPP_ERROR(get_logger(), "Invalid input data layout. Output may be incorrect");
  }
```

### Definition of done

1. list functions which make strong assumption on the data layout (e.g. `reinterpret_cast<PointXYZI*>`)
2. for each, add small check and report an error if the assumptions is wrong (the code is most likely buggy)

amadeuszsz, 2024-07-31:
Hi @VRichardJP 
Does recent point types migration covers your idea? https://github.com/autowarefoundation/autoware.universe/pull/6996
I see multiple check for point types are now performed in filters nodes.

VRichardJP, 2024-07-31:
I think so. 
The other day I updated autoware a saw a few pointcloud format error messages popping at runtime. This is very convenient as I could quickly find the root cause of in my sensor kit :)

closed 2024-07-31 by VRichardJP