autowarefoundation/autoware_universe #3386: DevOps Dojo: ROS Node Launch

kasperornmeck, 2023-04-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

DevOps: ROS Node Launch will address the topic of how ROS nodes are launched. Guidelines, documentation and changes to configuration files will be made, based on best-practices from cloud-native and software defined development methodologies.

Phase 1 - Motivation
- Add link

Phase 2 - Definition
- Add link

Phase 3 - Execution
- Add link

Phase 4 - Integration
- Add link

### Purpose

Phase 1 - Motivation:
- clear purpose
  - ...
- outline proposed work item
  - ...
- identify beneficiaries, stakeholders and shareholders
  - Autoware developers
    - consistent and rigid launch files
  - Users who deploy Autoware
    - ease of changing the launch pipeline to suit user's need

### Possible approaches

Involve the Autoware community to understand current deficiencies.

### Definition of done

Phase 1 is considered complete once satisfactory motivation has been outlined.

kasperornmeck, 2023-04-13:
Find discussion here: https://github.com/orgs/autowarefoundation/discussions/3425

kaspermeck-arm, 2023-10-12:
Fatih is working on a launch unifier:
* https://github.com/orgs/autowarefoundation/discussions/3881