autowarefoundation/autoware_universe #3389: Development plan: dynamic obstacle avoidance

takayuki5168, 2023-04-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, we have an avoidance module in behavior_path_planner only for static obstacles.
To drive safer without less risk of collision, avoidance of dynamic obstacles is required.

### Purpose

Avoid dynamic obstacles which are close to the ego or ego's trajectory to reduce the risk of collision.

### Possible approaches

The current algorithm for static obstacle avoidance will not work for dynamic obstacle avoidance.

The algorithm of path planning for static obstacle avoidance is as follows.
- calculate waypoints a certain lateral distance far from the obstacle
- connect waypoints and the current ego's lane's centerline smoothly

Connecting points smoothly does not work well in a complicated scene since it is a rule-based algorithm.
- avoid obstacles on the right and left side by driving like S-shape.
- avoid an obstacle during a lane change.

Dynamic obstacle avoidance requires a dynamic change of the waypoints.
This is the reason the current algorithm cannot be applied to dynamic obstacle avoidance.

---

Based on a quick discussion at TIER IV, by using the algorithm of the sideshift module in behavior_path_planner, we can realize dynamic obstacle avoidance.

The sideshift module plans a path from the lateral distance to shift and lateral jerk/acceleration limitation.

- determine dynamic obstacles to avoid
    - The position is close to the ego or ego's trajectory
    - TTC(time to collision) is low.
    - It will not go into the ego's lane by lane change.
    - Speed is high enough.
    - etc.
- calculate a lateral distance to shift to keep a space both from the dynamic obstacle and road boundary
- plan a path based on the  sideshift module

### Definition of done

Resolve remaining issues.

takayuki5168, 2023-04-28:
Current progress.
The parameters are tuned for the video to show the performance easily to see.
https://user-images.githubusercontent.com/20228327/235217296-52ce9757-9e98-4c4a-a190-73ed8a619d9b.mp4

yukkysaito, 2023-04-28:
This is a very nice feature! :+1:

takayuki5168, 2024-01-18:
Examples in the real world

https://github.com/autowarefoundation/autoware.universe/assets/20228327/403ce379-0178-49d0-a5fe-8c7787665022

https://github.com/autowarefoundation/autoware.universe/assets/20228327/f6995443-8f45-4678-9816-243f2a9d3e48

https://github.com/autowarefoundation/autoware.universe/assets/20228327/dd9b2021-e7cb-42bc-9ec8-b59b8eb66de9

[Screencast from 2024年01月16日 20時16分28秒.webm](https://github.com/autowarefoundation/autoware.universe/assets/20228327/1b49e8f5-eb63-4130-9035-1da9cda67d78)

https://github.com/autowarefoundation/autoware.universe/assets/20228327/3e1c9785-123d-4c9c-b496-256b6e235db1