autowarefoundation/autoware_universe #3526: Gear is not changed to PARK on goal anymore.

VRichardJP, 2023-04-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

During a recent test drive, I observed Autoware does not change the vehicle gear to PARK anymore upon reaching goal. Normally, this piece of code is supposed to do so when `ARRIVED_GOAL` state is received:
https://github.com/autowarefoundation/autoware.universe/blob/001a865bdae040784d3392677cb6d18c82309aac/control/shift_decider/src/shift_decider.cpp#L73-L74

Upon arrival, Autoware state changes quickly from `DRIVING` to `ARRIVED_GOAL`, to `FINALIZING`, then back to `WAITING_FOR_ROUTE`. Autoware is always in Autonomous mode and the vehicle is always engaged.

### Expected behavior

Gear is changed to park upon arrival if `park_on_goal_` is `true`

### Actual behavior

Gear unchanged.

### Steps to reproduce

Drive autonomously to any goal.

brkay54, 2023-05-11:
I couldn't observe the problem at commit sha in the universe: `f627fc30a6730af52348e0605437d7ed6806170e`

Could you recheck whether the issue still occurs or not?

brkay54, 2023-05-11:
Sorry for my mistake, it still occurs. I am going to work on this.

closed 2023-05-17 by tkimura4