autowarefoundation/autoware_universe #3548: map4_localization_launch: Cannot locate rosdep definition for [eagleye_rt]

TUMGEKA, 2023-04-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am trying to build a dewvelop docker using the provided autoware build.sh but I am getting a dependency error. 

### Expected behavior

Should build the docker image.

### Actual behavior

> #12 2584.5 Add distro "melodic"
#12 2584.5 Add distro "noetic"
#12 2584.5 Add distro "rolling"
#12 2584.5 updated cache in /root/.ros/rosdep/sources.cache
#12 2585.3 ERROR: the following packages/stacks could not have their rosdep keys resolved
#12 2585.3 to system dependencies:
#12 2585.3 map4_localization_launch: Cannot locate rosdep definition for [eagleye_rt]
#12 ERROR: process "/bin/bash -o pipefail -c ./setup-dev-env.sh -y $SETUP_ARGS universe   && pip uninstall -y ansible ansible-core   && mkdir src   && vcs import src < autoware.repos   && rosdep update   && DEBIAN_FRONTEND=noninteractive rosdep install -y --ignore-src --from-paths src --rosdistro \"$ROS_DISTRO\"   && apt-get clean   && rm -rf /var/lib/apt/lists/*" did not complete successfully: exit code: 1
------

### Steps to reproduce

1. `./docker/build.sh --platform linux/amd64 --no-prebuilt`

### Versions

- OS: Ubuntu 22.04
- ROS2 Humble
- Autoware Main

kminoda, 2023-04-26:
Is your Autoware up-to-date? Please follow [this update instruction](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/#how-to-update-a-workspace) carefully, and let me know if the issue still persists :+1:

(A new package was added here: https://github.com/autowarefoundation/autoware/pull/3261)

BonoloAWF, 2023-06-19:
@TUMGEKA if you still need help please create a Q&A question [here](https://github.com/orgs/autowarefoundation/discussions/categories/q-a)

closed 2023-06-19 by BonoloAWF