autowarefoundation/autoware_universe #3568: [obstacle avoidance planner]drivable area error, trajectory disconnects with vehicle body

ahuazuipiaoliang, 2023-04-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

drivable area looks odd when vehicle avoids obstacles. The reference trajectory disconnected with the vehicle body.

### Expected behavior

1. The drivable area should not be a odd polygon in the picture below.
2. The reference should be connected with the vehicle body.

[image]

### Actual behavior

In the picture above, the 2 lanelets were parallel and in opposite direction, which share a common boundary, because we created them using copy lanelet in Tier IV VMB. There was a U turn at the end of the down lanelet where the vehicle in, the U turn connected with the up lanelet.

### Steps to reproduce

in simple planning simulator
1. set the vehicle in the down lanelet
```yaml
header:
  stamp:
    sec: 1682646747
    nanosec: 563960181
  frame_id: map
pose:
  pose:
    position:
      x: -76.65474700927734
      y: 116.4011459350586
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.01176164365075141
```
2. set a goal pose in the up lanelet
```yaml
header:
  stamp:
    sec: 1682646776
    nanosec: 650720080
  frame_id: map
pose:
  position:
    x: -65.8907699584961
    y: 122.6998062133789
    z: -1.7958868438885511
  orientation:
    x: 0.0
    y: 0.0
    z: 0.9998301774823879
    w: -0.018428678616139057
```
3. set a static dumpy car that the vehicle could avoid 

### Versions

- OS: Ubuntu 22.04
- ROS2: Humble
- Autoware.universe: https://github.com/autowarefoundation/autoware.universe/commit/eeaff1521d9697e65a41a3561460280b1a0749b0
- Map: [map download](https://drive.google.com/drive/folders/1_aI2wDkTEyoyTD8hp10CkU66yvgZIM5E?usp=sharing)

### Possible causes

These 2 lanelets share the same linestring as the boundary. the order of points to generate the drivable area could make mistake in this situation.

takayuki5168, 2023-05-21:
@ahuazuipiaoliang Recently, some bug fix for generating the drivable area was merged. Could you try again with the latest main branch?

ahuazuipiaoliang, 2023-05-22:
@takayuki5168 I still meet some issues about planning in the newest version. Please have a look.
https://github.com/autowarefoundation/autoware.universe/issues/3774