autowarefoundation/autoware_universe #3569: LiDAR pointcloud disconnects with vehicle body

ahuazuipiaoliang, 2023-04-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The LiDAR pointcloud are not connected with the vehicle when the vehicle is moving.
The vehicle(orange box) should be at the center of rings of pointcloud.
It seems like the TF publishing is a little bit lagging.

[image]

### Expected behavior

The vehicle(orange box) should be at the center of rings of pointcloud.

### Actual behavior

The vehicle(orange box) was not at the center of rings of pointcloud.

### Steps to reproduce

1. launch the Autoware
2. drive the vehicle manually

### Versions

- OS: Ubuntu 22.04
- ROS: Humble
- Autoware.universe: https://github.com/autowarefoundation/autoware.universe/commit/eeaff1521d9697e65a41a3561460280b1a0749b0
- LiDAR: Robosense Helio 16P
- CPU: Intel 9 th, i9 9900k
- memory: 32GB
- GPU: Nvidia GeForce RTX3060

xmfcx, 2023-05-02:
Can you share the following to help us understand the issue better:

## Average available memory

While the Autoware is running, run following in a terminal (you can copy paste it and just run it):

It will sample last 10 seconds and print an average of the total available memory.
```bash
total_mem=0
samples=10

for i in $(seq 1 $samples); do
    mem=$(free -g | awk 'NR == 2 {print $4}')
    total_mem=$((total_mem + mem))
    sleep 1
done

average_mem=$(echo "scale=2; $total_mem / $samples" | bc)
echo "Average available memory: $average_mem GB"
```

It should the available memory like:
`Average available memory: 46.00 GB`

## Average cpu usage

Run `sar -u 1 5`. It will collect CPU usage statistics every 1 second for a duration of 5 seconds using the `sar` command.
We'd like to look at the `%idle` column average.

## Component container usage

Run `ros2 component list` and share the result with us.

Detailed info:
- https://docs.ros.org/en/humble/Concepts/About-Composition.html
- https://docs.ros.org/en/humble/Tutorials/Intermediate/Composition.html

## ROS2 communication delays

If CPU and memory are not strained, then we need to look into what might be causing the issue.

Backtrack:

- You will need to start tracking from the final topic (ndt aligned point cloud topic). 
  - And run:
    - `ros2 topic delay your_topic_name`
    - note the value
- run `ros2 topic info -v your_topic_name` 
  - to get the name of the node that publishes this topic.
  - the node name is the concatenation of `Node namespace` + `Node name`
- run `ros2 node info the_node_name`
  - look into the Subscribers and repeat the same steps for the same steps
  - look for the messages that might cause the problem
    - for the case of ndt, it's likely the point cloud input

In the end it is necessary to debug the source of the bad performance and this method may help.

@YamatoAndo do you have any suggestions?

YamatoAndo, 2023-05-08:
@ahuazuipiaoliang This behavior is indeed correct. 
The LiDAR point cloud being displayed is not the most recent one but rather the one acquired in the past.
It takes some time, approximately 100ms to acquire LiDAR packets and an additional few tens to a few hundred milliseconds for pointcloud_preprocess, before the LiDAR point cloud is displayed in RViz.
As a result, the LiDAR point cloud is centered around the past vehicle position.

YamatoAndo, 2023-05-08:
By the way, I don't know about Robosense Helio 16P as I have never used it, but please note that if there is no timestamp attached to each LiDAR point cloud, [distortion_corrector](https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/pointcloud_preprocessor/docs/distortion-corrector.md) may not work and Scan Matching accuracy may be decreased.

ahuazuipiaoliang, 2023-05-09:
> Can you share the following to help us understand the issue better:
> 
> ## Average available memory
> While the Autoware is running, run following in a terminal (you can copy paste it and just run it):
> 
> It will sample last 10 seconds and print an average of the total available memory.
> 
> ```shell
> total_mem=0
> samples=10
> 
> for i in $(seq 1 $samples); do
>     mem=$(free -g | awk 'NR == 2 {print $4}')
>     total_mem=$((total_mem + mem))
>     sleep 1
> done
> 
> average_mem=$(echo "scale=2; $total_mem / $samples" | bc)
> echo "Average available memory: $average_mem GB"
> ```
> 
> It should the available memory like: `Average available memory: 46.00 GB`
> 
> ## Average cpu usage
> Run `sar -u 1 5`. It will collect CPU usage statistics every 1 second for a duration of 5 seconds using the `sar` command. We'd like to look at the `%idle` column average.
> 
> ## Component container usage
> Run `ros2 component list` and share the result with us.
> 
> Detailed info:
> 
> * https://docs.ros.org/en/humble/Concepts/About-Composition.html
> * https://docs.ros.org/en/humble/Tutorials/Intermediate/Composition.html
> 
> ## ROS2 communication delays
> If CPU and memory are not strained, then we need to look into what might be causing the issue.
> 
> Backtrack:
> 
> * You will need to start tracking from the final topic (ndt aligned point cloud topic).
>   
>   * And run:
>     
>     * `ros2 topic delay your_topic_name`
>     * note the value
> * run `ros2 topic info -v your_topic_name`
>   
>   * to get the name of the node that publishes this topic.
>   * the node name is the concatenation of `Node namespace` + `Node name`
> * run `ros2 node info the_node_name`
>   
>   * look into the Subscribers and repeat the same steps for the same steps
>   * look for the messages that might cause the problem
>     
>     * for the case of ndt, it's likely the point cloud input
> 
> In the end it is necessary to debug the source of the bad performance and this method may help.
> 
> @YamatoAndo do you have any suggestions?

Thanks, I will follow the instructions to collect the runtime information.

ahuazuipiaoliang, 2023-05-09:
> By the way, I don't know about Robosense Helio 16P as I have never used it, but please note that if there is no timestamp attached to each LiDAR point cloud, [...]

Thanks for your reply, I think it was not correct, because I've never met such issue in ROS1 when I use some pointcloud preprocessors.

kminoda, 2023-05-10:
Maybe it would also be good to try running rosbag with lower rate, e.g. `ros2 bag play YOURBAG -r 0.1`

xmfcx, 2023-05-10:
@ahuazuipiaoliang also I've created following performance diagnosis and troubleshooting section:

https://autowarefoundation.github.io/autoware-documentation/pr-365/support/troubleshooting/performance-troubleshooting/#performance-troubleshooting

It helped someone else's similar problems, it might help with yours too.

Make sure to check all the steps.

ahuazuipiaoliang, 2023-05-11:
@xmfcx thanks for you advice

ahuazuipiaoliang, 2023-05-11:
@xmfcx I tried to tune the DDS setting, the pointcloud lagging problem has been optimized.

xmfcx, 2023-05-11:
@ahuazuipiaoliang It's great to hear that!

I'm closing this issue, if you have problems, don't hesitate to reopen it :)

closed 2023-05-11 by xmfcx