autowarefoundation/autoware_universe #3585: Installation problem in "Docker installation for development"

yuntianyi-chen, 2023-04-29:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I followed the instructions in [Docker installation](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation-devel/). Every step is correct, but when I executed the command, some errors occurred. 

[image]
[image]

- Error log below:

```
Ign:12 http://deb.debian.org/debian stretch-updates/main all Packages
Ign:9 http://deb.debian.org/debian stretch/main all Packages
Ign:10 http://deb.debian.org/debian stretch/main amd64 Packages
Ign:11 http://deb.debian.org/debian stretch-updates/main amd64 Packages
Ign:12 http://deb.debian.org/debian stretch-updates/main all Packages
Ign:9 http://deb.debian.org/debian stretch/main all Packages
Ign:10 http://deb.debian.org/debian stretch/main amd64 Packages
Ign:11 http://deb.debian.org/debian stretch-updates/main amd64 Packages
Ign:12 http://deb.debian.org/debian stretch-updates/main all Packages
Ign:9 http://deb.debian.org/debian stretch/main all Packages
Err:10 http://deb.debian.org/debian stretch/main amd64 Packages
  404  Not Found
Err:11 http://deb.debian.org/debian stretch-updates/main amd64 Packages
  404  Not Found
Ign:12 http://deb.debian.org/debian stretch-updates/main all Packages
Ign:3 http://security.debian.org/debian-security stretch/updates/main all Packages
Ign:4 http://security.debian.org/debian-security stretch/updates/main amd64 Packages
Ign:3 http://security.debian.org/debian-security stretch/updates/main all Packages
Ign:4 http://security.debian.org/debian-security stretch/updates/main amd64 Packages
Ign:3 http://security.debian.org/debian-security stretch/updates/main all Packages
Ign:4 http://security.debian.org/debian-security stretch/updates/main amd64 Packages
Ign:3 http://security.debian.org/debian-security stretch/updates/main all Packages
Ign:4 http://security.debian.org/debian-security stretch/updates/main amd64 Packages
Ign:3 http://security.debian.org/debian-security stretch/updates/main all Packages
Err:4 http://security.debian.org/debian-security stretch/updates/main amd64 Packages
  404  Not Found
Reading package lists...
W: The repository 'http://security.debian.org/debian-security stretch/updates Release' does not have a Release file.
W: The repository 'http://deb.debian.org/debian stretch Release' does not have a Release file.
W: The repository 'http://deb.debian.org/debian stretch-updates Release' does not have a Release file.
E: Failed to fetch http://security.debian.org/debian-security/dists/stretch/updates/main/binary-amd64/Packages  404  Not Found
E: Failed to fetch http://deb.debian.org/debian/dists/stretch/main/binary-amd64/Packages  404  Not Found
E: Failed to fetch http://deb.debian.org/debian/dists/stretch-updates/main/binary-amd64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.

Removing intermediate container 455d186b6ae2
no more output and success not detected
Failed to build detector image
WARNING unable to detect os for base image 'ghcr.io/autowarefoundation/autoware-universe:latest-cuda', maybe the base image does not exist

```

### Expected behavior

I should be able to enter the docker container without errors. In fact, several weeks ago, I could successfully install Autoware through docker. But after I upgraded my OS to Ubuntu22.04, it fails to install through docker. By the way, source installation is feasible.

### Actual behavior

I followed the instructions in [Docker installation](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation-devel/).

### Steps to reproduce

Follow every step in the [Docker installation](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/docker-installation-devel/).

### Versions

- OS: Ubuntu 22.04
- Autoware.universe

### Possible causes

I guess some dependencies in Docker are no longer accessible.

BonoloAWF, 2023-04-29:
Does this solve your problem https://github.com/orgs/autowarefoundation/discussions/3470 ?

yuntianyi-chen, 2023-04-29:
> Does this solve your problem https://github.com/orgs/autowarefoundation/discussions/3470 ?

Thank you!  It solved my problem. I guess some links of dependency were outdated recently.

closed 2023-04-29 by yuntianyi-chen