autowarefoundation/autoware_universe #3628: Improve MPC modeling accuracy

TakaHoribe, 2023-05-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

The current mpc model uses the "error kinematics model in Frenet coordinate". This model obtains the reference path information from only curvature and indirectly regenerates the reference path information by integrating curvature in the MPC process. However, curvature calculation could have non-small errors, and the reference path information given by integration could also have errors.

In this video, the green path is the trajectory given by the planning module, and the white path is the trajectory re-generated in the MPC model from curvature. There is a large difference when driving in a curve.

https://user-images.githubusercontent.com/21360593/236382083-79c2a5ce-58d0-4958-8d9e-5499385f420d.mp4

### Purpose

To improve the MPC tracking accuracy

### Possible approaches

improve model

### Definition of done

mpc performance is measured and improved

TakaHoribe, 2023-05-06:
In the video above, the white line is generated by the linearized model, while in the video below, the nonlinear path model is calculated.
- orange-line: nonlinear path model generated from curvature
- white-line: predicted ego-path of the MPC

The accuracy is much better in the nonlinear model.

https://user-images.githubusercontent.com/21360593/236626809-1f7cff17-f66b-46a0-9f73-c452045d05ec.mp4

VRichardJP, 2023-05-10:
Silly question: why does MPC needs to regenerate the reference path?

13304490790, 2023-06-05:
Hello,I have a question why there some difference when calculate discrete dynamic model .