autowarefoundation/autoware_universe #3673: ground remover package cannot delete ghost points appears front of car when passing speedbump

Zeysthingz, 2023-05-11:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently we are using ```scan_ground_filter``` algorithm in ground segmentation  node but when passing speed bumps  the algorithm can not work efficiently because vehicle is shaken and can not delete all ground points ghost points occurs  front of vehicle.Because of point occurs on the way ```obstacle stop planner``` stops the car. Here is  image an example of this problem:
[image]

Default parameter values are below:
```
  global_slope_max_angle_deg: 10.0
  local_slope_max_angle_deg: 13.0 # recommended 30.0 for non elevation_grid_mode
  split_height_distance: 0.2
  non_ground_height_threshold: 0.20
  grid_mode_switch_radius: 20.0
  gnd_grid_buffer_size: 4
  detection_range_z_max: 2.5
  elevation_grid_mode: true
```

Mainly  increasing   ```non_ground_height_threshold```  parameter affects  deleting these points and this parameter is height threshold of non ground objects for elevation grid mode.Currently we tuned parameters and took better result but when increase parameters to much  we are losing to much points.

Here is tuned parameters below:
```
    global_slope_max_angle_deg 50.0
    local_slope_max_angle_deg 35.0 
    split_height_distance 0.25
    non_ground_height_threshold 0.50
    grid_mode_switch_radius 30.0
    gnd_grid_buffer_size 1
    detection_range_z_max 2.5
    elevation_grid_mode true
```
Here is a bag file to help understand problem [ground_remover_bag](https://drive.google.com/drive/folders/1Z5qhQCc0ZsYNf1DM9GbC7n6JBQsndPg9)

### Purpose

To improve algorithm when passing speed bumps 

### Possible approaches

improve algorithm 

### Definition of done

Tuned current algorithm and tested it on vehicle

Zeysthingz, 2023-06-02:
Here is a bag file also includes localization topics.
[bag_includes_imu](https://drive.google.com/drive/folders/1x5LpzwIPwes2_QxdmLCFUCbcw_rMN2_e)

kaancolak, 2023-06-13:
I take over this issue

amadeuszsz, 2024-07-05:
@vividf friendly ping

vividf, 2024-07-05:
This issue should solved by using the latest universe using 3d distortion correction in the distortion corrector node. 

- https://github.com/autowarefoundation/autoware.universe/issues/6657#event-13377100753
- https://github.com/autowarefoundation/autoware.universe/pull/7137#event-13376124349

closed 2024-07-05 by vividf