autowarefoundation/autoware_universe #3682: Reduce warning messages from tier4_autoware_utils::transformPointCloud

rej55, 2023-05-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

If we run planning simulator, `tier4_autoware_utils::transformPointCloud` outputs a lot of warning messages.

If it outputs a lot of warning messages, it is difficult to notify any other important error or warning, so I want to reduce it.

### Purpose

To make it easier to see any other warnings.

### Possible approaches

Remove the following lines:
https://github.com/kaancolak/autoware.universe/blob/e76e5b6af13437309c449c3aae718c05b317155c/common/tier4_autoware_utils/include/tier4_autoware_utils/transform/transforms.hpp#L32
https://github.com/kaancolak/autoware.universe/blob/e76e5b6af13437309c449c3aae718c05b317155c/common/tier4_autoware_utils/include/tier4_autoware_utils/transform/transforms.hpp#L44

### Definition of done

Warning messages from `tier4_autoware_utils::transformPointCloud` is reduced.

rej55, 2023-05-12:
@kaancolak cc @takayuki5168 
Hello.

I want to reduce the warning messages from `tier4_autoware_utils::transformPointCloud`.
I think it is enough to output warning messages from the node using this utility, so it is no problem to remove the warning message in `tier4_autoware_utils::transformPointCloud`.

How do you think?

mitsudome-r, 2023-05-16:
https://github.com/autowarefoundation/autoware.universe/pull/3708#issuecomment-1547092243

> If it is spamming the terminal, have we considered using RCLCPP_WARN_THROTTLE?

closed 2024-02-29 by rej55