autowarefoundation/autoware_universe #3693: Adding Orthometric height as an option to the gnss poser package

meliketanrikulu, 2023-05-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

We work with different sensors. Some sensors give the height value as ellipsoid height, while others give it as orthometric height. Gnss poser currently uses ellipsoid height. We think that orthometric height should also be added as an option. otherwise, there is a height difference between the gnss poser output using the height from gnss and our map, and we have a problem while initializing it. 

### Purpose

To be able to use the gnss poser correctly at different height types while working with different sensors.

### Possible approaches

GNSS POSER : 
input: ellipsoid height 
output: orthometric height

GNSS POSER FUTURE REQUEST :
input: ellipsoid height /  orthometric height (optional)
output:  orthometric height

### Definition of done

We expect gnss poser to be usable with both height types

YamatoAndo, 2023-05-15:
Thank you!
This feature is something we have been wanting too!

farukbaykara, 2023-05-25:
I tested the feature on Robione vehicle in campus. Initialized immediately without problem at different locations.

closed 2023-05-31 by YamatoAndo