autowarefoundation/autoware_universe #3724: Use pluginlib for behavior_velocity_planner

isamu-takagi, 2023-05-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

The behavior_velocity_planner package contains many scene modules. I plan to use pluginlib to split the scene module into separate packages. This allows you to put additional scene modules in separate repositories. In addition, since the number of dependent libraries is minimized, the build efficiency is improved.

### Purpose

- Improve scene module extensibility.
- Improve build efficiency.

### Possible approaches

Move the interface and utilities into the common package.
Move each scene module into its own package.

[image]

### Definition of done

- [ ] use pluginlib

BonoloAWF, 2023-05-22:
@isamu-takagi can this issue be closed since https://github.com/autowarefoundation/autoware.universe/pull/3734 is merged?

isamu-takagi, 2023-05-23:
The tasks listed in ”definition of done" section have not yet been completed.

isamu-takagi, 2023-06-22:
I decided that I needed a wrapper class to use the constructor.

closed 2023-06-22 by isamu-takagi