autowarefoundation/autoware_universe #3725: Create a release/2023.04 branch

esteve, 2023-05-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

The [autoware.repos file for release/2023.04](https://github.com/autowarefoundation/autoware/blob/release/2023.04/autoware.repos#L27) points to a commit in the `autoware.universe` repo that fails to build because it's missing https://github.com/autowarefoundation/autoware.universe/pull/3280

### Purpose

Be able to build Debian packages for the 2023.04 release

### Possible approaches

Create a new branch in the autoware.universe repository (e.g. `release/2023.04`) which can incorporate fixes and have the `autoware.repos` file in https://github.com/autowarefoundation/autoware/blob/release/2023.04/autoware.repos#L27 point to the `release/2023.04` branch for `autoware.universe`

### Definition of done

Debian packages for the 2023.04 release can be built as part of the GitHub CI workflow.

BonoloAWF, 2023-05-16:
Will this be a stable release which incorporates all the new features from Bus ODD? Several community members have requested this.

esteve, 2023-05-18:
@BonoloAWF this would only just for hotfixes (basically build issues), no features. These monthly releases are just snapshots of what's in `main` at the end of the month,  they are not meant as milestones.

esteve, 2024-03-27:
Closing as this is a very old issue.

closed 2024-03-27 by esteve