autowarefoundation/autoware_universe #3769: delete and create virtual_wall_marker from behavior_velocity_planner's scene module approprately

kyoichi-sugahara, 2023-05-19:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

`virtual_wall_marker` pubslihed from `behavior_velocity_planner` is deleted by setting the lifetime, for example, to 0.2 seconds and each module runs at 10Hz, the wall marker will be displayed unnecessarily for 0.1 seconds after it is no longer needed. 
Although it may seem like a short time when considering real time, when running a heavy simulator at a slower speed, for example, at 1/10 of the normal speed, the marker will be displayed for 1 second despite being unnecessary, which can confuse developers.

For these reasons, I believe that an deletion function at every cycle is necessary when modules determine that the marker is no longer needed.

### Purpose

Ensure that the `virtual_wall_marker` published from the scene module of `behavior_velocity_planner` can be properly deleted/created at the right timing.

### Possible approaches

A point to be cautious about is that when deleting or creating `virtual_wall_marker', it is necessary to specify the namespace and id. However, if the same scene module is launched (e.g., when there are multiple intersections, multiple intersection modules can be launched), `virtual_wall_marker` that should be generated in different locations may be mistakenly deleted due to duplication of namespaces and id. Therefore, when the same scene module is launched multiple times, it is necessary to give a different namespace as an argument. As proposed in this [PR](https://github.com/autowarefoundation/autoware.universe/pull/3654), by adding the module_id as a prefix to the namespace, we need to ensure that the namespaces do not duplicate.

Another point to note is that when using the module_id as a prefix to the namespace, it is necessary for the module_id to be unique among the same scene modules.

### Definition of done

Make sure that each scene module of the `behavior_velocity_planner` can use `VirtualWallMarkerCreator` to properly delete/create `virtual_wall_markers` at the right timing.

- [x] intersection
- [x] crosswalk
- [ ] blind_spot(module_id should be change to unique)
- [ ] no_stopping_area
- [x] detection_area
- [ ] occulusion_spot(module_id should be change to unique)
- [ ] out_of_lane
- [ ] run_out
- [ ] speed_bump
- [x] stop_line
- [ ] traffic_light
- [ ] virtual_traffic_light(module_id should be change to unique)

kyoichi-sugahara, 2023-05-29:
In the following PR, we(@kyoichi-sugahara @soblin @maxime-clem )made changes to ensure that the `virtual_wall_marker` can be created/deleted in each scene module.

https://github.com/autowarefoundation/autoware.universe/pull/3833
https://github.com/autowarefoundation/autoware.universe/pull/3836
https://github.com/autowarefoundation/autoware.universe/pull/3832
https://github.com/autowarefoundation/autoware.universe/pull/3837
https://github.com/autowarefoundation/autoware.universe/pull/3838
https://github.com/autowarefoundation/autoware.universe/pull/3835
https://github.com/autowarefoundation/autoware.universe/pull/3834
https://github.com/autowarefoundation/autoware.universe/pull/3839

maxime-clem, 2023-07-31:
@kyoichi-sugahara can we close this issue ?

kyoichi-sugahara, 2023-07-31:
@maxime-clem 
Yes!
Thank you for contribute to this issue a lot!!!

closed 2023-07-31 by kyoichi-sugahara