autowarefoundation/autoware_universe #3774: [Planning] cannot planing trajectory using my map

ahuazuipiaoliang, 2023-05-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The Autoware works fine with the map in [sample-map-planning tutorial](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/)
[image]

but cannot plan the trajectory using my lanelet2 map.
[image]

You can download my map for testing:

[pointcloud_map](https://drive.google.com/file/d/1xY3yM7uRYtNanQNhipON-2FIuKmtSZDB/view?usp=share_link)
[lanelet2_map](https://drive.google.com/file/d/118K4ZHrk9huNWMdkeRyJmvga0gJuQCVX/view?usp=sharing)

### Expected behavior

should send the global path and local path

### Actual behavior

Got an error from mission planner
[image]

### Steps to reproduce

1. launch autoware with simple simulator
2. set initial pose
3. set goal pose

### Versions

- OS: Ubuntu 22.04
- ROS: Humble
- Autoware.universe: https://github.com/autowarefoundation/autoware.universe/commit/ffc8b712c5bb440b5a19dbd9fe818b1d087fdde8

ahuazuipiaoliang, 2023-05-20:
After downgrading to the [autoware.unierse-release/2023.04](https://github.com/autowarefoundation/autoware/tree/autoware.universe-release/2023.04), it works

TakaHoribe, 2023-05-25:
@ahuazuipiaoliang The warning on your terminal "Goal is not valid! Please check position and angle of goal_pose" appears, for example, when the goal position is not in the lanelet, or the goal posture is opposite to the lanelet direction. ([code is here](https://github.com/tier4/autoware.universe/blob/4ec79136e/planning/mission_planner/src/lanelet2_plugins/default_planner.cpp/#L423-L427))

In that case, the goal sent by user is not accepted by the mission_planner.

You can try with a larger angle threshold (e.g. 180 deg) set in yaml file [here](https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/config/planning/mission_planning/mission_planner/mission_planner.param.yaml/#L7).

TakaHoribe, 2023-05-25:
@mitsudome-r It is sometimes difficult for users to see why mission_planner rejects the goal. What do you think about adding a parameter to enable/disable the goal validation?

TakaHoribe, 2023-05-25:
I couldn't load your map correctly. Do we have a special setup to load the map (changing the origin, etc)?

[image]

ahuazuipiaoliang, 2023-05-25:
> I couldn't load your map correctly. Do we have a special setup to load the map (changing the origin, etc)?
> 
> [image]

I think you should set the coordinate format as `local` in the config file of lanelet2_map_loader.

ahuazuipiaoliang, 2023-05-25:
@TakaHoribe My goal pose was correct, since I have try Autoware for a long time and I can't set correct goal in elder version of Autwoare.universe

TakaHoribe, 2023-05-25:
@ahuazuipiaoliang I checked it in my environment, and it seems the lane boundary and center line are broken on Rviz. But the Lanelet looks ok in the vectormap builder.

@mitsudome-r Any idea?

[image]

ahuazuipiaoliang, 2023-05-25:
@TakaHoribe That's so weird cause I have successfully run elder version of Autoware.universe with my more complicated maps, but after I upgraded universe, I could not use my map. But the newest version of universe is compatible with the sample-map that provided by AWF. I thinks it is a little bit tricky to make useful vector map for newest version of Autoware.univers.

TakaHoribe, 2023-05-25:
Yes, it can be a bug. Let me investigate more.

mitsudome-r, 2023-05-25:
@ahuazuipiaoliang 
I think the issue is caused by the following:
- the map has wrong MGRS code annotated in the map. (The lanelet map you provided crosses over multiple MGRS grid which is very unlikely so I'm assuming that it is annotated incorrectly)
- That is why you have to specify "local" for projection type to use the local_x, local_y tags in the lanelet2 map instead.
- However, there seems to be a bug in "local" projection type in the universe code (and the bug only happens when you have invalid MGRS grid annotated in the map)

I can create the fix PR in map_loader package to fix the bug. In the meantime, you can try using the lanelet2 map that I attached, which has MGRS grid annotation removed so that it doesn't trigger the bug existing in the current code.

[lanelet2_map.zip](https://github.com/autowarefoundation/autoware.universe/files/11566528/lanelet2_map.zip)

mitsudome-r, 2023-05-25:
Once this PR is merged, you should be able to drive using the original map as well. https://github.com/autowarefoundation/autoware.universe/pull/3825

ahuazuipiaoliang, 2023-05-26:
@mitsudome-r Thanks for your reply. In my using cases, I don't have MGRS grid reference in the vector because I can provide the lat/lon/alt origin to the gnss_poser for pose initialization, and the coordinates of vector map is aligned to the coordinates of pointcloud map.
BTW, what is a valid MGRS grid?

mitsudome-r, 2023-05-26:
Sorry, "valid" was not the correct phrase, and maybe "correct" MGRS grid is more accurate expression.

By looking at tags in lanelet2 map, I believe you created the map using TIER IV's vector map builder(VMB) by loading point cloud map and drawing lanes on it. When you started creating a lanelet2 map, I think it asked you to specify the MGRS grid of your map. If your point cloud is created with MGRS coordinates, then you should specify MGRS grid value of your map. If your point cloud map is in local coordinates(like in your case) then you should keep it the default "99XXX" as MGRS grid to tell VMB that your map will be in local coordinates. 
[image]

However, when I checked your original map, it had 17TLG specified, which isn't true. This has caused some issue around projection at load time in Autoware causing your issue
[image]
.

mitsudome-r, 2023-05-26:
@ahuazuipiaoliang 
Just confirm, were you able to drive correctly with the map I posted in https://github.com/autowarefoundation/autoware.universe/issues/3774#issuecomment-1563088650?

ahuazuipiaoliang, 2023-05-29:
@mitsudome-r I try try this branch today with my map, thanks a lot for your support

ahuazuipiaoliang, 2023-05-29:
@mitsudome-r thanks, it works now.

closed 2023-05-29 by mitsudome-r