autowarefoundation/autoware_universe #3882: map_loader and map_tf_generator can not load georeferenced maps

enesadastec, 2023-06-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

"map_loader" and "map_tf_generator" packages can not load maps which have UTM coordinates like 4.000.000. Because these packages are reading ".pcd" files as pcl::PointXYZ which only supports float32 precision coordinates.

If we try to load UTM maps like that, the loaded map looks like corrupted on RVIZ (because all points are rounded to float32)
[image]

Here is an example pcd file which is not compatible with pcl::PointXYZ (float32)
[image]

Discussion topic:
https://github.com/orgs/autowarefoundation/discussions/3355

### Purpose

Loading UTM maps which are not compatible with pcl::PointXYZ (float32)

### Possible approaches

- Set an origin point from the map and then substract this origin point from all points while reading map process (convert georeferenced coordinates to local coordinates)

### Definition of done

- Converting UTM maps to local coordinates and be able to visualize on RVIZ

kminoda, 2023-06-02:
TIER IV is currently working on a very initial investigation for addressing maps that overlap MGRS grids or UTM zones. One possible approach we are considering is to create original projection from WGS84 to local coordinate. I believe this will subsidiary solve the issue mentioned here, as long as the map size are within appropriate range (say, 0km to 100km for x and y). With this update, **Autoware will be able to run without any worries on overlapping MGRS or UTM grids**.

We are currently in the final step of internal design review for this topic, so I think we can share the design within this month (or maybe in two weeks, if everything goes well). 

Here are some questions from our side:
- (Just as a confirmation) Does the above approach would solve your issue, doesn't it?
- What is your progress and future plan for the above issue?

We are happy to collaborate with you if necessary :+1:

enesadastec, 2023-06-05:
Hi,

Thanks for your comment and brief information about your investigation.

1) As you can see on the photo in first comment, my map's X coordinate is 700.000 and Y coordinate is 4.700.000. Im not sure if your solution can handle situations like that.

2) Im currently aware of this problem and made a presentation in Mapping WG. As a solution, I was planning to add a feature that substracts origin coordinates that defined by user from all cloud points (with this way convert UTM coordinates to local coordinates).

If you currently working on this problem, we can wait for your test results to check whether this problem is solved with your solution. Then, if the problem is still occurs, we can reinvestigate the problem and we can talk about possible other solutions.

Also I will be very happy to contribute with you if there is something I can do.

kminoda, 2023-06-06:
@enesadastec Thank you for the comments!  Let me move to the above discussion :pray:

kminoda, 2023-06-21:
@enesadastec Sorry I forgot to post it here. Here's my proposal. Any comments or feedbacks are welcome!
https://github.com/orgs/autowarefoundation/discussions/3582

enesadastec, 2023-06-21:
> 

Thanks for sharing the link. I checked the proposal and I believe it will solve this problem.

kminoda, 2023-11-22:
@enesadastec Hi, I believe this issue is now solved with the proposal https://github.com/orgs/autowarefoundation/discussions/3582. Let us know if you have any further questions, and please close the issue if it is solved from your end as well.

kminoda, 2023-11-29:
Since this is staled, I will convert this as closed. Please feel free to reopen this issue if you have any further issues. Thank you for your understanding.

closed 2023-11-29 by kminoda