autowarefoundation/autoware_universe #3896: distortion_corrector should update "distance" field

VRichardJP, 2023-06-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Because LIDAR sensors do not shoot laser simultaneously in all directions, raw sensor clouds look distorted when the vehicle turns or drive fast. `pointcloud_preprocessor::DistortionCorrectorComponent` use vehicle twist/imu data to rectify that distortion. Basically, it interpolates cloud points xyz values 
 to make it as if all point laser had been shot simultaneously.

In the implementation `x`, `y` and `z` fields are modified, but not `azimuth` nor `distance`. Downstream nodes thus cannot use these 2 fields reliably.

### Expected behavior

`pointcloud_preprocessor::DistortionCorrectorComponent` also rectifies `azimuth` and `distance`

### Actual behavior

Does not 

### Steps to reproduce

N/A

vividf, 2024-07-12:
The PR for fixing this is blocked because Velodyne's output azimuth is already wrong before the distortion corrector node.

vividf, 2024-09-05:
This issue can be solved by this PR https://github.com/autowarefoundation/autoware.universe/pull/8380

closed 2024-10-28 by vividf