autowarefoundation/autoware_universe #3912: Unable to re-build pointcloud_preprocessor in Docker

jamestkpoon, 2023-06-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am trying to investigate some issues in the concatenate_data component, and wanted to debug it.
However I find that I cannot re-build the pointcloud_preprocessor package.
Part of my Dockerfile is as follows:

```
ARG AUTOWARE_BASE_TAG=humble-latest-prebuilt-amd64
FROM ghcr.io/autowarefoundation/autoware-universe:${AUTOWARE_BASE_TAG}

# pull autoware src for development/debugging
ENV AUTOWARE_WS_DIR /autoware
RUN wget https://raw.githubusercontent.com/autowarefoundation/autoware/main/autoware.repos -O /tmp/autoware.repos
RUN mkdir -p ${AUTOWARE_WS_DIR}/src
RUN vcs import ${AUTOWARE_WS_DIR}/src < /tmp/autoware.repos
RUN vcs pull ${AUTOWARE_WS_DIR}/src
```

After building with the default ```AUTOWARE_BASE_TAG```, inside the container I run:
```
cd /autoware
colcon build --packages-select pointcloud_preprocessor
```

The build quickly reaches 94% progress, and then seems to get stuck forever, e.g:
```
[8min 56.8s] [0/1 complete] [pointcloud_preprocessor:build 94% - 8min 56.2s]
```

Any suggestions would be appreciated.

### Expected behavior

pointcloud_preprocessor package builds successfully.

### Actual behavior

Build gets stuck.

### Steps to reproduce

1. Build Docker image using partial Dockerfile above
2. ```
   cd /autoware
   colcon build --packages-select pointcloud_preprocessor
   ```

### Versions

Host OS: Ubuntu 22.04. Hardware resources should be adequate (i7, 16 GB RAM).
Docker container is run with ```--memory 12g --memory-swap 20g --cpus 8```

### Possible causes

Not sure, the pointcloud_preprocessor CMakeLists.txt seems quite normal.

greensky0802, 2023-11-07:
It seems likely to be caused by insufficient memory,preferably 32g of RAM

mitsudome-r, 2026-03-24:
I'm closing the issue that was inactive for a long time. Feel free to reopen if the issue is still relevant.