autowarefoundation/autoware_universe #3940: Map Loader on universe cannot load custom .pcd and .lanelet2 map on source and docker installations

yaukaizhi, 2023-06-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Unable to load custom map (pcd and lanelet2) on docker devel image on AGX Xavier (ARM64):
- Able to load planning tutorial map
- Unable to load rosbag tutorial map sometimes
- Blank RViz screen when loading own map (by changing the map path)

Unable to load custom map (pcd and lanelet2) on source install on x86:
- Able to load planning tutorial map
- Able to load rosbag tutorial map and rosbag play
- Blank RViz screen when loading own map (by changing the map path)

However, prebuilt image was able to load all maps just by changing the map path

### Expected behavior

Able to load all maps by changing the map path

### Actual behavior

Unable to load custom map on docker devel and source install, only works on prebuilt image.

### Steps to reproduce

source ~/autoware/install/setup.bash

ros2 launch autoware_launch planning_simulator.launch.xml map_path:=/path_to_your_map vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit

maxime-clem, 2023-06-12:
Can you make sure if this is a visualization issue ?
Please run the following two commands after starting Autoware.
```
ros2 topic echo /map/pointcloud_map 
ros2 topic echo /map/vector_map
```
If these commands show some message, then the maps are correctly published and the problem may come from rviz.

yaukaizhi, 2023-06-12:
Ahh, I did not set the target frame to map, however the lanelet2 map seems to be distorted and extends beyond the pcd map. Would you be able to help with this?

[image]

maxime-clem, 2023-06-12:
You can make sure the pointcloud and lanelet map are correctly aligned with this tool: https://tools.tier4.jp/vector_map_builder_ll2/ (you need to make a free account if you do not already have one).

yaukaizhi, 2023-06-12:
Yes they are aligned in vector map builder, but not aligned in RViz
[image]

maxime-clem, 2023-06-12:
Have you tried changing the `lanelet2_map_projector_type` parameter to `local` ?
https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/config/map/lanelet2_map_loader.param.yaml#L3

If this does not work please open a new issue for the map being misaligned.

yaukaizhi, 2023-06-12:
yes that worked! turns out i was editing the wrong yaml file, I was editing the params in launcher/autoware_launch/autoware_launch/config/map/lanelet2_map_loader.param.yaml

Thank you!