autowarefoundation/autoware_universe #3975: refactor(autoware_auto_mapping_msgs): replace autoware_auto_mapping_msgs with autoware_map_msgs

cyn-liu, 2023-06-14:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Replace all usages of
- https://github.com/tier4/autoware_auto_msgs/tree/tier4/main/autoware_auto_mapping_msgs/msg
with
- https://github.com/autowarefoundation/autoware_msgs/blob/main/autoware_map_msgs/msg

Mainly replacing 
- https://github.com/tier4/autoware_auto_msgs/blob/tier4/main/autoware_auto_mapping_msgs/msg/HADMapBin.idl
with 
- https://github.com/autowarefoundation/autoware_msgs/blob/main/autoware_map_msgs/msg/LaneletMapBin.msg

```bash
$ grep -ril "autoware_auto_mapping_msgs::" ./
./core/external/autoware_auto_msgs/autoware_auto_mapping_msgs/srv/HADMapService.idl
./core/external/autoware_auto_msgs/autoware_auto_mapping_msgs/msg/HADMapSegment.idl
./core/external/autoware_auto_msgs/autoware_auto_planning_msgs/msg/HADMapRoute.idl
./core/external/autoware_auto_msgs/autoware_auto_planning_msgs/msg/Route.idl
./core/autoware_common/tmp/lanelet2_extension/include/lanelet2_extension/utility/route_checker.hpp
./core/autoware_common/tmp/lanelet2_extension/include/lanelet2_extension/utility/message_conversion.hpp
./core/autoware_common/tmp/lanelet2_extension/README.md
./core/autoware_common/tmp/lanelet2_extension/lib/message_conversion.cpp
./core/autoware_common/tmp/lanelet2_extension/test/src/test_message_conversion.cpp
./core/autoware_common/tmp/lanelet2_extension/test/src/test_route_checker.cpp
./universe/autoware.universe/sensing/pointcloud_preprocessor/include/pointcloud_preprocessor/vector_map_filter/vector_map_inside_area_filter.hpp
./universe/autoware.universe/sensing/pointcloud_preprocessor/include/pointcloud_preprocessor/vector_map_filter/lanelet2_map_filter_nodelet.hpp
./universe/autoware.universe/sensing/pointcloud_preprocessor/docs/vector-map-filter.md
./universe/autoware.universe/sensing/pointcloud_preprocessor/docs/vector-map-inside-area-filter.md
./universe/autoware.universe/sensing/pointcloud_preprocessor/src/vector_map_filter/lanelet2_map_filter_nodelet.cpp
./universe/autoware.universe/sensing/pointcloud_preprocessor/src/vector_map_filter/vector_map_inside_area_filter.cpp
./universe/autoware.universe/common/tier4_traffic_light_rviz_plugin/src/traffic_light_publish_panel.hpp
./universe/autoware.universe/planning/mission_planner/include/mission_planner/mission_planner_plugin.hpp
./universe/autoware.universe/planning/mission_planner/src/mission_planner/mission_planner.hpp
./universe/autoware.universe/planning/scenario_selector/include/scenario_selector/scenario_selector_node.hpp
./universe/autoware.universe/planning/scenario_selector/README.md
./universe/autoware.universe/planning/scenario_selector/src/scenario_selector_node/scenario_selector_node.cpp
./universe/autoware.universe/planning/behavior_velocity_planner/README.md
./universe/autoware.universe/planning/behavior_velocity_planner/src/node.hpp
./universe/autoware.universe/planning/behavior_velocity_planner/src/node.cpp
./universe/autoware.universe/planning/costmap_generator/include/costmap_generator/costmap_generator.hpp
./universe/autoware.universe/planning/costmap_generator/README.md
./universe/autoware.universe/planning/costmap_generator/nodes/costmap_generator/costmap_generator_node.cpp
./universe/autoware.universe/planning/route_handler/include/route_handler/route_handler.hpp
./universe/autoware.universe/planning/obstacle_stop_planner/README.md
./universe/autoware.universe/planning/planning_test_utils/include/planning_interface_test_manager/planning_interface_test_manager.hpp
./universe/autoware.universe/planning/planning_test_utils/include/planning_interface_test_manager/planning_interface_test_manager_utils.hpp
./universe/autoware.universe/planning/obstacle_velocity_limiter/include/obstacle_velocity_limiter/obstacle_velocity_limiter_node.hpp
./universe/autoware.universe/planning/obstacle_velocity_limiter/src/obstacle_velocity_limiter_node.cpp
./universe/autoware.universe/planning/behavior_path_planner/include/behavior_path_planner/behavior_path_planner_node.hpp
./universe/autoware.universe/planning/behavior_path_planner/README.md
./universe/autoware.universe/planning/behavior_path_planner/test/test_drivable_area_expansion.cpp
./universe/autoware.universe/planning/static_centerline_optimizer/include/static_centerline_optimizer/type_alias.hpp
./universe/autoware.universe/map/map_loader/include/map_loader/lanelet2_map_loader_node.hpp
./universe/autoware.universe/map/map_loader/include/map_loader/lanelet2_map_visualization_node.hpp
./universe/autoware.universe/map/map_loader/src/lanelet2_map_loader/lanelet2_map_visualization_node.cpp
./universe/autoware.universe/map/map_tf_generator/Readme.md
./universe/autoware.universe/map/map_tf_generator/src/vector_map_tf_generator_node.cpp
./universe/autoware.universe/control/lane_departure_checker/include/lane_departure_checker/lane_departure_checker_node.hpp
./universe/autoware.universe/control/lane_departure_checker/README.md
./universe/autoware.universe/perception/crosswalk_traffic_light_estimator/include/crosswalk_traffic_light_estimator/node.hpp
./universe/autoware.universe/perception/crosswalk_traffic_light_estimator/README.md
./universe/autoware.universe/perception/traffic_light_selector/src/traffic_light_selector.hpp
./universe/autoware.universe/perception/detected_object_validation/include/detected_object_filter/object_lanelet_filter.hpp
./universe/autoware.universe/perception/detected_object_validation/object-lanelet-filter.md
./universe/autoware.universe/perception/detected_object_validation/src/object_lanelet_filter.cpp
./universe/autoware.universe/perception/traffic_light_map_based_detector/include/traffic_light_map_based_detector/node.hpp
./universe/autoware.universe/perception/traffic_light_map_based_detector/README.md
./universe/autoware.universe/perception/traffic_light_map_based_detector/src/node.cpp
./universe/autoware.universe/perception/elevation_map_loader/include/elevation_map_loader/elevation_map_loader_node.hpp
./universe/autoware.universe/perception/elevation_map_loader/README.md
./universe/autoware.universe/perception/elevation_map_loader/src/elevation_map_loader_node.cpp
./universe/autoware.universe/perception/traffic_light_visualization/include/traffic_light_map_visualizer/node.hpp
./universe/autoware.universe/perception/traffic_light_visualization/README.md
./universe/autoware.universe/perception/traffic_light_visualization/src/traffic_light_map_visualizer/node.cpp
./universe/autoware.universe/perception/map_based_prediction/include/map_based_prediction/map_based_prediction_node.hpp
./universe/autoware.universe/perception/map_based_prediction/README.md
./universe/autoware.universe/perception/traffic_light_arbiter/include/traffic_light_arbiter/traffic_light_arbiter.hpp
./universe/autoware.universe/perception/traffic_light_arbiter/README.md

```

### Purpose

To deprecate the `autoware_auto_mapping_msgs` and switch to autoware_map_msgs.

### Possible approaches

I probably have to modify all of these files in a single PR.

### Definition of done

doing.
cc. @x

cyn-liu, 2023-08-11:
@xmfcx

xmfcx, 2023-08-11:
> I have a question about the following pair, their data types are inconsistent, do they represent the same data?
> `uint8 map_format` -> `string name_map`

No, In the old format, it was possible to have multiple map representations like `MAP_FORMAT_LANELET2`, `MAP_FORMAT_OPEN_DRIVE`

https://github.com/tier4/autoware_auto_msgs/blob/e1795354161ed54d66659366262578863bd2b862/autoware_auto_mapping_msgs/msg/HADMapBin.idl#L5C1-L7C7

But in the new format, by the name, it is assumed to be Lanelet2. The field is removed.

closed 2024-10-15 by cyn-liu