autowarefoundation/autoware_universe #3993: `map_height_fitter` fails to detect map ground

VRichardJP, 2023-06-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In a simulated environment, the `map_height_fitter` fails to locate the PCD map ground and the initial position ends up 150 meters under the map (z=0):

[image]

[image]

In the log, I see:
```
[rviz2-69] [INFO 1686888780.878679225] [rviz2]: Setting estimate pose: Frame:map, Position(52451.3, 65997.6, 0), Orientation(0, 0, -0.833062, 0.55318) = Angle: -1.96924 (operator()() at ./src/main.cpp:62)
[initial_pose_adaptor-66] [INFO 1686888780.879371161] [default_ad_api.helpers.initial_pose_adaptor]: map fit1: 52451.301 65997.586 0.000 (fit() at /home/sig/autoware/src/universe/autoware.universe/map/map_height_fitter/src/map_height_fitter.cpp:169)
[initial_pose_adaptor-66] [INFO 1686888780.879819454] [default_ad_api.helpers.initial_pose_adaptor]: Send request to map_loader (get_partial_point_cloud_map() at /home/sig/autoware/src/universe/autoware.universe/map/map_height_fitter/src/map_height_fitter.cpp:103)
[initial_pose_adaptor-66] [INFO 1686888780.881185460] [default_ad_api.helpers.initial_pose_adaptor]: Loaded partial pcd map from map_loader (grid size: 0) (get_partial_point_cloud_map() at /home/sig/autoware/src/universe/autoware.universe/map/map_height_fitter/src/map_height_fitter.cpp:115)
[initial_pose_adaptor-66] Failed to find match for field 'x'.
[initial_pose_adaptor-66] Failed to find match for field 'y'.
[initial_pose_adaptor-66] Failed to find match for field 'z'.
[initial_pose_adaptor-66] [INFO 1686888780.881249434] [default_ad_api.helpers.initial_pose_adaptor]: map fit2: 52451.301 65997.586 0.000 (fit() at /home/sig/autoware/src/universe/autoware.universe/map/map_height_fitter/src/map_height_fitter.cpp:189)
```

I don't know if it is related, but PCL complains it cannot find xyz fields. Lokking at the code, it looks like the error comes from here:

https://github.com/autowarefoundation/autoware.universe/blob/2465bbfa1de791794b2f49088ef301e20ad7e568/map/map_height_fitter/src/map_height_fitter.cpp#L120-L133 

As you can see, the `pcd_msg` message is created from the concatenation of all partials pointclouds, but its `fields` are never set. Maybe it causes `pcl::fromROSMsg()` to fail and return an empty cloud?

### Expected behavior

ground height correctly detected

### Actual behavior

ground height not detected

### Steps to reproduce

N/A

VRichardJP, 2023-06-16:
I think there is a design issue with the `pointcloud_map_loader`:
- `enable_partial_load` is set to `true` by default in the `autoware_launch`
- if this flag is set, then `map_height_fitter` call the service to replace the current map
- yet, if the flag is set but only 1 pcd map is used, the service returns an empty cloud.

Considering that using only one PCD map should be the default use case, I don't think the current behaviour is correct.

kminoda, 2023-06-26:
Thank you for the report! We encountered a similar issue before, which I believe was resolved with this PR: https://github.com/autowarefoundation/autoware.universe/pull/3863. Would you check if the PR is incorporated in your Autoware?

VRichardJP, 2023-06-26:
@kminoda Unfortunatelly, I just pulled latest autoware and lost track of the exact hash I was working on, but I see my previous pull was on June 16, so after your PR has been merged. I also remember finding the cyclinder overlap function as I inspected the issue.

I fixed the issue by disabling the partial map loading (I have only 1 pcd file).

VRichardJP, 2023-06-27:
The problem still occurs:
[image]

The initial pose is exactly at z=0 in map frame, while the actual points are ~150 meters above.

kminoda, 2023-06-27:
Hmm, I see no problem in my Autoware (also latest). I translated a `sample-map-rosbag` from tutorial with `pcl_transform_point_cloud pointcloud_map.pcd output.pcd -trans 0.0,0.0,200.0`, load it in Autoware and run with the tutorial rosbag. It starts successfully with z-axis also aligned.

What kind of map are you using? Is it your own map? It does not include any noise points especially around z=0, right?

[image]

VRichardJP, 2023-06-27:
With the dynamic map loading disabled I have no problem, even if I move the map up as you did:
[image]

However if dynamic map loading is enabled (the default autoware_launch), the map does not load at all. Localization is never initialized.
[image]

In the logs I see this:
```
[ndt_scan_matcher-31] [INFO 1687854665.894965925] [localization.pose_estimator.ndt_scan_matcher]: Waiting for pcd loader service. Check if the enable_differential_load in pointcloud_map_loader is set `true`. (MapUpdateModule() at /home/sig/autoware/src/universe/autoware.universe/localization/ndt_scan_matcher/src/map_update_module.cpp:73)
[component_container_mt-1] [INFO 1687854665.905057830] [perception.object_recognition.detection.voxel_based_compare_map_filter]: service not available, waiting again ... (VoxelGridDynamicMapLoader() at /home/sig/autoware/src/universe/autoware.universe/perception/compare_map_segmentation/src/voxel_grid_map_loader.cpp:306)
[rviz2-68] Warning: Invalid frame ID "viewer" passed to canTransform argument source_frame - frame does not exist
[rviz2-68]          at line 93 in ./src/buffer_core.cpp
[rviz2-68] Warning: Invalid frame ID "viewer" passed to canTransform argument source_frame - frame does not exist
[rviz2-68]          at line 93 in ./src/buffer_core.cpp
[rviz2-68] Warning: Invalid frame ID "viewer" passed to canTransform argument source_frame - frame does not exist
[rviz2-68]          at line 93 in ./src/buffer_core.cpp
[rviz2-68] Warning: Invalid frame ID "viewer" passed to canTransform argument source_frame - frame does not exist
[rviz2-68]          at line 93 in ./src/buffer_core.cpp
[rviz2-68] Warning: Invalid frame ID "viewer" passed to canTransform argument source_frame - frame does not exist
[rviz2-68]          at line 93 in ./src/buffer_core.cpp
[rviz2-68] Warning: Invalid frame ID "viewer" passed to canTransform argument source_frame - frame does not exist
[rviz2-68]          at line 93 in ./src/buffer_core.cpp
[rviz2-68] Warning: Invalid frame ID "viewer" passed to canTransform argument source_frame - frame does not exist
[rviz2-68]          at line 93 in ./src/buffer_core.cpp
[rviz2-68] Warning: Invalid frame ID "viewer" passed to canTransform argument source_frame - frame does not exist
[rviz2-68]          at line 93 in ./src/buffer_core.cpp
[rviz2-68] Warning: Invalid frame ID "viewer" passed to canTransform argument source_frame - frame does not exist
[rviz2-68]          at line 93 in ./src/buffer_core.cpp
[rviz2-68] Warning: Invalid frame ID "viewer" passed to canTransform argument source_frame - frame does not exist
[rviz2-68]          at line 93 in ./src/buffer_core.cpp
[ekf_localizer-33] [WARN 1687854666.327254766] [localization.pose_twist_fusion_filter.ekf_localizer]: The node is not activated. Provide initial pose to pose_initializer (warnThrottle() at /home/sig/autoware/src/universe/autoware.universe/localization/ekf_localizer/include/ekf_localizer/warning.hpp:34)
[rviz2-68] Warning: Invalid frame ID "viewer" passed to canTransform argument source_frame - frame does not exist
[rviz2-68]          at line 93 in ./src/buffer_core.cpp
[rviz2-68] Warning: Invalid frame ID "viewer" passed to canTransform argument source_frame - frame does not exist
[rviz2-68]          at line 93 in ./src/buffer_core.cpp
```

Looks like the pcd loader service is not started properly.

kminoda, 2023-06-27:
Sorry for repeating it again, but could you double check that your Autoware is updated to the latest version properly with clean build? I will do it for my environment as well as soon as my other task is done.

Also, could you provide the log around
- ndt_scan_matcher
- map_loader
- map_height_fitter
- pose_initializer
- ekf_localizer

by e.g. doing `ros2 launch autoware_launch ... | grep -e ndt_scan_matcher -e map_loader ...` ?

BTW at least from the log you've provided at the top of this issue,
```
[rviz2-69] [INFO 1686888780.878679225] [rviz2]: Setting estimate pose: Frame:map, Position(52451.3, 65997.6, 0), Orientation(0, 0, -0.833062, 0.55318) = Angle: -1.96924 (operator()() at ./src/main.cpp:62)
[initial_pose_adaptor-66] [INFO 1686888780.879371161] [default_ad_api.helpers.initial_pose_adaptor]: map fit1: 52451.301 65997.586 0.000 (fit() at /home/sig/autoware/src/universe/autoware.universe/map/map_height_fitter/src/map_height_fitter.cpp:169)
[initial_pose_adaptor-66] [INFO 1686888780.879819454] [default_ad_api.helpers.initial_pose_adaptor]: Send request to map_loader (get_partial_point_cloud_map() at /home/sig/autoware/src/universe/autoware.universe/map/map_height_fitter/src/map_height_fitter.cpp:103)
[initial_pose_adaptor-66] [INFO 1686888780.881185460] [default_ad_api.helpers.initial_pose_adaptor]: Loaded partial pcd map from map_loader (grid size: 0) (get_partial_point_cloud_map() at /home/sig/autoware/src/universe/autoware.universe/map/map_height_fitter/src/map_height_fitter.cpp:115)
[initial_pose_adaptor-66] Failed to find match for field 'x'.
[initial_pose_adaptor-66] Failed to find match for field 'y'.
[initial_pose_adaptor-66] Failed to find match for field 'z'.
[initial_pose_adaptor-66] [INFO 1686888780.881249434] [default_ad_api.helpers.initial_pose_adaptor]: map fit2: 52451.301 65997.586 0.000 (fit() at /home/sig/autoware/src/universe/autoware.universe/map/map_height_fitter/src/map_height_fitter.cpp:189)
```

the map_loader seems to have started successfully, but just to load the target pointcloud given the query.

VRichardJP, 2023-06-27:
I just pulled and recompiled a clean autoware code:
```
=== src/core/autoware.core (git) ===
commit 99891401473b5740e640f5a0cc0412c0984b8e0b (HEAD -> main, origin/main, origin/HEAD, humble_with_targets)
Author: awf-autoware-bot[bot] <94889083+awf-autoware-bot[bot]@users.noreply.github.com>

    chore: sync files (#78)

commit 794aed1b9dd5eaf5e1ce349fb96ac7adb87a8f9a
Author: awf-autoware-bot[bot] <94889083+awf-autoware-bot[bot]@users.noreply.github.com>

    chore: sync files (#76)

commit 0a3421239fcf576d3e73036c6b5da1daf2b8c8d0
Author: awf-autoware-bot[bot] <94889083+awf-autoware-bot[bot]@users.noreply.github.com>

    chore: sync files (#75)
=== src/core/autoware_adapi_msgs (git) ===
commit 7cd80335ad1a4edc1955de5062dd5a4bd9d4de7b (HEAD -> main, origin/main, origin/HEAD)
Author: Kah Hooi Tan <41041286+tkhmy@users.noreply.github.com>

    feat(autoware_adapi_v1_msgs): add vehicle status msgs (#24)

commit 6de4fe5858d96b2e787847b29c36858e06ae3c98
Author: awf-autoware-bot[bot] <94889083+awf-autoware-bot[bot]@users.noreply.github.com>

    chore: sync files (#37)

commit 8d0fc25179d62f9de6d3ec235f652b42e3c0dc11
Author: Kosuke Takeuchi <kosuke.tnp@gmail.com>

    refactor(start_planner): rename pull out to start planner (#36)
=== src/core/autoware_common (git) ===
commit 536b628d6cc87ddf877efb351ff9d38f12fec620 (HEAD -> main, origin/main, origin/HEAD)
Author: Kosuke Takeuchi <kosuke.tnp@gmail.com>

    feat(lanelet2_extension): move combineLanelets unnamed namespace (#193)

commit d65a73812d63f0ddf6134e07b58d351c626dfa19
Author: Ahmed Ebrahim <36835765+ahmeddesokyebrahim@users.noreply.github.com>

    refactor(no_drivable_lane): updating no_drivable_lane doc link after refactoring updates (#188)

commit 268a8c84c6de0e864fd159f15fae03beb04f0e93
Author: Kenji Miyake <31987104+kenji-miyake@users.noreply.github.com>

    ci(pre-commit): add check-jsonschema (#189)
=== src/core/autoware_msgs (git) ===
commit 52a856d29601860413e4261079e0bf9205c737d1 (HEAD -> main, origin/main, origin/HEAD)
Author: Mingyu1991 <115005477+Mingyu1991@users.noreply.github.com>

    revert: "fix(autoware_perception_msgs): add Header to tlr messages and change TrafficLightElement.msg status" (#69)

commit 18588df6eb7e7f694bd725bebd44f569f616964f
Author: Mingyu1991 <115005477+Mingyu1991@users.noreply.github.com>

    fix(autoware_perception_msgs): add Header to tlr messages and change TrafficLightElement.msg status (#68)

commit 76d12470ca1788108a9507cd854e64f5d02e6c7c
Author: Mingyu1991 <115005477+Mingyu1991@users.noreply.github.com>

    feat(autoware_perception_msgs): add UP_LEFT_ARROW and UP_RIGHT_ARROW to traffic_light_element (#67)
=== src/core/external/autoware_auto_msgs (git) ===
commit e1795354161ed54d66659366262578863bd2b862 (HEAD -> tier4/main, origin/tier4/main, origin/HEAD)
Author: Mingyu1991 <115005477+Mingyu1991@users.noreply.github.com>

    add UP_LEFT_ARROW and UP_RIGHT_ARROW for traffic_light (#42)

commit f6642370c6f4f42a5dc0b6c1fc4a21396d4dc34c (humble_with_targets)
Author: Esteve Fernandez <esteve@apache.org>
[... 345 lines cut ...]
[component_container_mt-1] [0m[INFO 1687861372.376411255] [perception.object_recognition.detection.voxel_based_compare_map_filter]: service not available, waiting again ... (VoxelGridDynamicMapLoader() at /home/sig/autoware/src/universe/autoware.universe/perception/compare_map_segmentation/src/voxel_grid_map_loader.cpp:306)[0m
[ndt_scan_matcher-31] [0m[INFO 1687861372.400041912] [localization.pose_estimator.ndt_scan_matcher]: Waiting for pcd loader service. Check if the enable_differential_load in pointcloud_map_loader is set `true`. (MapUpdateModule() at /home/sig/autoware/src/universe/autoware.universe/localization/ndt_scan_matcher/src/map_update_module.cpp:73)[0m
[component_container_mt-1] [0m[INFO 1687861373.376609396] [perception.object_recognition.detection.voxel_based_compare_map_filter]: service not available, waiting again ... (VoxelGridDynamicMapLoader() at /home/sig/autoware/src/universe/autoware.universe/perception/compare_map_segmentation/src/voxel_grid_map_loader.cpp:306)[0m
[ndt_scan_matcher-31] [0m[INFO 1687861373.400213846] [localization.pose_estimator.ndt_scan_matcher]: Waiting for pcd loader service. Check if the enable_differential_load in pointcloud_map_loader is set `true`. (MapUpdateModule() at /home/sig/autoware/src/universe/autoware.universe/localization/ndt_scan_matcher/src/map_update_module.cpp:73)[0m
[component_container_mt-1] [0m[INFO 1687861374.376877243] [perception.object_recognition.detection.voxel_based_compare_map_filter]: service not available, waiting again ... (VoxelGridDynamicMapLoader() at /home/sig/autoware/src/universe/autoware.universe/perception/compare_map_segmentation/src/voxel_grid_map_loader.cpp:306)[0m
[ndt_scan_matcher-31] [0m[INFO 1687861374.400552815] [localization.pose_estimator.ndt_scan_matcher]: Waiting for pcd loader service. Check if the enable_differential_load in pointcloud_map_loader is set `true`. (MapUpdateModule() at /home/sig/autoware/src/universe/autoware.universe/localization/ndt_scan_matcher/src/map_update_module.cpp:73)[0m
[component_container_mt-1] [0m[INFO 1687861375.378281922] [perception.object_recognition.detection.voxel_based_compare_map_filter]: service not available, waiting again ... (VoxelGridDynamicMapLoader() at /home/sig/autoware/src/universe/autoware.universe/perception/compare_map_segmentation/src/voxel_grid_map_loader.cpp:306)[0m
[... cut ...] /compare_map_segmentation/src/voxel_grid_map_loader.cpp:306)[0m
[ndt_scan_matcher-31] [0m[INFO 1687861382.402719313] [localization.pose_estimator.ndt_scan_matcher]: Waiting for pcd loader service. Check if the enable_differential_load in pointcloud_map_loader is set `true`. (MapUpdateModule() at /home/sig/autoware/src/universe/autoware.universe/localization/ndt_scan_matcher/src/map_update_module.cpp:73)[0m
```

I then tried with the dynamic map loading disabled, and it just works as expected.

kminoda, 2023-06-28:
Thank you. But I still have no clue where the issue comes from, so let me ask some more question:

1. What is the output of `ros2 node info /map/pointcloud_map_loader` in that situation?
2. What does the `/map/pointcloud_map_loader` output in `autoware_launch` terminal? You can see that by e.g. simply launching autoware_launch with `map:=true localization:=false planning:=false perception:=false sensing:=false ...` and see the outputs.

kminoda, 2023-06-28:
BTW I also updated my Autoware up-to-date and still fails to reproduce the error (so as one of my colleagues)

VRichardJP, 2023-06-28:
I observe something strange:
- if I start the `logging_simulator.launch.xml`, the lanelet2 and PCD maps do not appear in Rviz
- but if I only start the `map` module, then the maps are visible in Rviz. 

In the first case, the `pointcloud_map_loader` is node is simply never loaded. Even after a few minutes have passed, I get this:
```
$ ros2 node list | grep /map
/map/lanelet2_map_loader
/map/map_container
/map/map_hash_generator
```
No wonder the service is not available.

I don't get this problem at all if only the `map` module is enabled:
```
$ ros2 node list | grep /map
/map/lanelet2_map_loader
/map/lanelet2_map_visualization
/map/map_container
/map/map_hash_generator
/map/pointcloud_map_loader
/map/vector_map_tf_generator
```

The problem is exactly the same with `autoware.launch.xml`: when all modules but `map` are disabled it works, but when everything is running it doesn't. It have tried to see if one particular module is responsible for this, but it does not look like it. At this point, it looks like black magic to me.

VRichardJP, 2023-06-28:
Using gdb I am trying to understand why the pointcloud map loader component is never started. The program gets stuck forever after loading the `Lanelet2MapLoaderNode` component:
```
Reading symbols from /opt/ros/humble/lib/rclcpp_components/component_container...
(No debugging symbols found in /opt/ros/humble/lib/rclcpp_components/component_container)
Starting program: /opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=map_container -r __ns:=/map -p use_sim_time:=True -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff6deb640 (LWP 55616)]
[New Thread 0x7ffff65ea640 (LWP 55617)]
[New Thread 0x7ffff5de9640 (LWP 55618)]
[New Thread 0x7ffff55e8640 (LWP 55619)]
[New Thread 0x7ffff4de7640 (LWP 55620)]
[New Thread 0x7fffe7fff640 (LWP 55621)]
[New Thread 0x7fffe77fe640 (LWP 55622)]
[New Thread 0x7fffe6ffd640 (LWP 55623)]
[New Thread 0x7fffe67fc640 (LWP 55624)]
[INFO 1687937239.099893814] [map.map_container]: Load Library: /home/sig/autoware/install/map_loader/lib/liblanelet2_map_loader_node.so (create_component_factory() at ./src/component_manager.cpp:114)
[INFO 1687937239.970020821] [map.map_container]: Found class: rclcpp_components::NodeFactoryTemplate<Lanelet2MapLoaderNode> (create_component_factory() at ./src/component_manager.cpp:127)
[INFO 1687937239.970099537] [map.map_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<Lanelet2MapLoaderNode> (create_component_factory() at ./src/component_manager.cpp:129)
[New Thread 0x7fffe5d2b640 (LWP 55643)]
```
If I interrupt the program, I can see ROS2 is stuck waiting for something on cycloneDDS side:
```
^C
Thread 1 "component_conta" received signal SIGINT, Interrupt.
__futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x555555589168) at ./nptl/futex-internal.c:57
57	./nptl/futex-internal.c: No such file or directory.
(gdb) bt
#0  __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x555555589168) at ./nptl/futex-internal.c:57
#1  __futex_abstimed_wait_common (cancel=true, private=0, abstime=0x0, clockid=0, expected=0, futex_word=0x555555589168) at ./nptl/futex-internal.c:87
#2  __GI___futex_abstimed_wait_cancelable64 (futex_word=futex_word@entry=0x555555589168, expected=expected@entry=0, clockid=clockid@entry=0, abstime=abstime@entry=0x0, private=private@entry=0) at ./nptl/futex-internal.c:139
#3  0x00007ffff7973ac1 in __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x555555589118, cond=0x555555589140) at ./nptl/pthread_cond_wait.c:503
#4  ___pthread_cond_wait (cond=0x555555589140, mutex=0x555555589118) at ./nptl/pthread_cond_wait.c:627
#5  0x00007ffff75461bd in ddsrt_cond_wait () from /opt/ros/humble/lib/x86_64-linux-gnu/libddsc.so.0
#6  0x00007ffff7546285 in ddsrt_cond_waituntil () from /opt/ros/humble/lib/x86_64-linux-gnu/libddsc.so.0
#7  0x00007ffff752f07d in ?? () from /opt/ros/humble/lib/x86_64-linux-gnu/libddsc.so.0
#8  0x00007ffff75da7cf in rmw_wait () from /opt/ros/humble/lib/librmw_cyclonedds_cpp.so
#9  0x00007ffff78a9708 in rcl_wait () from /opt/ros/humble/lib/librcl.so
#10 0x00007ffff7e6edbc in rclcpp::Executor::wait_for_work(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /opt/ros/humble/lib/librclcpp.so
#11 0x00007ffff7e71ab3 in rclcpp::Executor::get_next_executable(rclcpp::AnyExecutable&, std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /opt/ros/humble/lib/librclcpp.so
#12 0x00007ffff7e78f31 in rclcpp::executors::SingleThreadedExecutor::spin() () from /opt/ros/humble/lib/librclcpp.so
#13 0x00005555555565d7 in ?? ()
#14 0x00007ffff7909d90 in __libc_start_call_main (main=main@entry=0x555555556310, argc=argc@entry=28, argv=argv@entry=0x7fffffff3898) at ../sysdeps/nptl/libc_start_call_main.h:58
#15 0x00007ffff7909e40 in __libc_start_main_impl (main=0x555555556310, argc=28, argv=0x7fffffff3898, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffff3888) at ../csu/libc-start.c:392
#16 0x0000555555556775 in ?? ()
(gdb) 
```

Running `pointcloud_map_loader` component in its own container seems to fix the issue, however it does not make any sense:
```python
container = ComposableNodeContainer(
        name="map_container",
        namespace="",
        package="rclcpp_components",
        executable=LaunchConfiguration("container_executable"),
        composable_node_descriptions=[
            lanelet2_map_loader,
            lanelet2_map_visualization,
            # pointcloud_map_loader,
            map_tf_generator,
        ],
        output="screen",
    )
    
    container2 = ComposableNodeContainer(
        name="map_container2",
        namespace="",
        package="rclcpp_components",
        executable=LaunchConfiguration("container_executable"),
        composable_node_descriptions=[
            pointcloud_map_loader,
        ],
        prefix="gnome-terminal --",
        output="screen",
    )

    group = GroupAction(
        [
            PushRosNamespace("map"),
            container, container2,
            map_hash_generator,
        ]
    )
```

```
[... cut ...]
[INFO 1687938307.456190626] [map.pointcloud_map_loader]: Load ../data/autoware_data_universe/sample-map-rosbag//pointcloud_map.pcd (1 out of 1) (loadPCDFiles() at /home/sig/autoware/src/universe/autoware.universe/map/map_loader/src/pointcloud_map_loader/pointcloud_map_loader_module.cpp:77)
[INFO 1687938307.473006366] [map.pointcloud_map_loader]: Create PCD metadata, as the pointcloud is a single file. (getPCDMetadata() at /home/sig/autoware/src/universe/autoware.universe/map/map_loader/src/pointcloud_map_loader/pointcloud_map_loader_node.cpp:111)
```

kminoda, 2023-07-03:
I am sorry I may not be able to be your help, given that the issue cannot be reproduced on my (or my colleagues') environment... Let us know if you have any update on this or got some clue on it.

isouf, 2023-07-07:
Thank you for raising this issue @VRichardJP.

I think I experience the same problem when I run the [Planning Tutorial](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/#parking-scenario):

[image]

[image]

I am running the latest Autoware `main` version with Docker installation (`ghcr.io/autowarefoundation/autoware-universe:latest-cuda`) and NVIDIA GPU enabled.

kminoda, 2023-07-12:
@isouf I think that issue is irrelevant from the one @VRichardJP has reported. It's due to the [simple_planning_simulator](https://github.com/autowarefoundation/autoware.universe/blob/67c95e9d54783ff295040773feca2b88a6539abc/simulator/simple_planning_simulator/src/simple_planning_simulator/simple_planning_simulator_core.cpp#L450-L478) where it does not take into account the z position from `/initialpose3d`.

isouf, 2023-07-13:
> @isouf I think that issue is irrelevant from the one @VRichardJP has reported. It's due to the [...]

Thank you @kminoda for the answer. I was hopping that this issue is related to the fact that I am not able to set the `OperationMode` to `AUTO` despite having the `Routing=SET` and the `Localization=INITIALIZED`. As you can see in the images attached above, the `AUTO` button is (for some unknown reason) disabled.

kminoda, 2023-07-14:
@isouf Thank you for the clarification! And not sure if it's related to the z-axis issue. Maybe better to create another issue or discussion.

qianxingqx, 2023-09-04:
> 

Hello, I have also encountered this problem. How did you solve it? Thanks!

qianxingqx, 2023-09-04:
> Thank you @kminoda for the answer. I was hopping that this issue is related to the fact that I am not able to set the `OperationMode` to `AUTO` despite having the `Routing=SET` and the [...]

Hello, I have also encountered this problem. How did you solve it? Thank you!

qianxingqx, 2023-09-04:
> @isouf Thank you for the clarification! And not sure if it's related to the z-axis issue. Maybe better to create another issue or discussion.

Thank you for your answer. I have the same problem with @isouf . Do you have any suggestions for this issue?

kminoda, 2023-09-04:
@qianxingqx You mean about the planning simulator with the map from tutorial, right?
It works fine with the latest Autoware on my PC. If the issue persists, please provide us with your specific environment and way to reproduce. 
[image]

Also, since this thread is getting longer, it would be better to create another issue.

kminoda, 2023-11-29:
@VRichardJP Since this is staled, I would like to close this issue in a week if you don't encounter this issue any longer or not up-to-date. Please feel free to reopen this issue if you have any further issues. Thank you for your understanding.

VRichardJP, 2023-11-29:
No problem.

As I read back some of my comments, it could be related to https://github.com/ros2/rclcpp/issues/2362

closed 2023-11-29 by kminoda