autowarefoundation/autoware_universe #4004: IPlugin Tensor Deprecated, cannot colcon build

yaukaizhi, 2023-06-18:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

[image]

Received this error after git pull and rebuilding

### Expected behavior

successful build without errors

### Actual behavior

Received build errors

### Steps to reproduce

vcs pull
source /opt/ros/humble/setup.bash
rosdep update
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

maxime-clem, 2023-06-19:
The compilation issue may come from your CUDA installation. Which version do you have installed ?
Personally I have no issue with CUDA version 12.0, but the default version installed by the `setup-dev-env.sh` script is version 11.6 (https://github.com/autowarefoundation/autoware/blob/main/amd64.env#L6).

One temporary workaround you can try is to build with the colcon option `--cmake-args -DCMAKE_CXX_FLAGS='-w'` so that this error will only be treated as a warning. Otherwise you probably need to downgrade CUDA.

yaukaizhi, 2023-06-20:
Yep I had Cuda 12.1 and had to downgrade it back to 11.6 for it to work! Thanks!

closed 2023-06-20 by yaukaizhi