autowarefoundation/autoware_universe #4015: When the initial position of the vehicle and the current lane line are at a significant angle, there is a certain probability of abnormal steering of the vehicle.

lixianqiang, 2023-06-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When I select a initial pose and a goal pose on a map I have drawn myself, if the initial pose is set at a significant angle with the lane lines, the output steering angle signal from the controller will make the vehicle track towards the desired trajectory. However, the output steering command will continue to instruct the vehicle to turn, resulting in deviation from the desired path.

### Expected behavior

Due to the significant angular error between initial point and the reference lane lines, the controller will generate a large steering output signal, causing the vehicle to steer towards the direction of the lane line. As the error gradually decreases, the steering output decreases as well. Eventually, the vehicle aligns its heading with the lane line.

### Actual behavior

The steering output does not decrease as the angular error decreases. Instead, it continues to output steering signals, causing the vehicle to deviate from the lane lines.

### Steps to reproduce

1. modify the map config in autoware/src/launcher/autoware_launch/autoware_launch/config/map/lanelet2_map_loader.param.yaml, set MGRS to local in parameter lanelet2_map_projector_type
2. executor the follower command "ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/factory_guiyang_outside sensor_model:=sample_sensor_kit vehicle_model:=sample_vehicle"
3. setting the initial pose and goal pose through replay the rosbag.
4. click the 'Auto' button in rviz. i found that the car will deviate from the lane after running some time

### Versions

- autoware.universe: main

### Additional context

video:
[录屏 2023年06月19日 20时51分35秒.webm](https://github.com/autowarefoundation/autoware.universe/assets/24970103/3c677bb0-9958-418e-8de0-d5a1830809c2)
map file:
https://drive.google.com/drive/folders/16wSBrGN4aWGJonxda5pPHAn4dNwnpwnj?usp=sharing
ros bag:
https://drive.google.com/drive/folders/1DKcynZEF08vmBVIAs6XNyXlAL8tWsaJK?usp=drive_link

TakaHoribe, 2023-06-20:
@lixianqiang The predicted path of the MPC seems to be broken in this picture. It usually happens when the MPC optimization fails. Did you get the error messages about MPC like "optimization failed" on the terminal?

[image]

lixianqiang, 2023-06-20:
@TakaHoribe  Yes, the message "qp solver error" or "optimization failed. Stop MPC" appeared on the terminal.
[image]

TakaHoribe, 2023-06-20:
@lixianqiang Thank you. I guess it is due to the [recent change](https://github.com/autowarefoundation/autoware.universe/pull/3810) of the steering_rate_limit.

Would you test with the following parameters? It worked in my environment.

```
    steer_rate_lim_dps_list_by_curvature: [40.0, 50.0, 60.0]   
    curvature_list_for_steer_rate_lim: [0.001, 0.002, 0.01]      
    steer_rate_lim_dps_list_by_velocity: [60.0, 50.0, 40.0]      
    velocity_list_for_steer_rate_lim: [10.0, 15.0, 20.0]  
```

If this solves your issue, a potential issue would raise: "mpc optimization fails with strong steering rate limit".

lixianqiang, 2023-06-20:
The parameters your provided worked in my environment as well.

lixianqiang, 2023-06-20:
@TakaHoribe  The parameters your provided worked in my environment as well.

TakaHoribe, 2023-06-20:
@lixianqiang I created the PR to change the default parameter. At least this PR fixes this issue. Let me track the root cause in another issue.

https://github.com/autowarefoundation/autoware.universe/pull/4021
https://github.com/autowarefoundation/autoware_launch/pull/405

closed 2023-06-20 by lixianqiang

TakaHoribe, 2023-06-20:
@lixianqiang I created a new issue https://github.com/autowarefoundation/autoware.universe/issues/4024 to track this problem. Thank you!