autowarefoundation/autoware_universe #4024: MPC fails with a small steering rate limit

TakaHoribe, 2023-06-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When a small steering rate limit is set to the MPC, the optimization fails.

It is reported in the https://github.com/autowarefoundation/autoware.universe/issues/4015. Please check the details of the issue.

The previous issue was addressed by modifying the default parameters. However, the fundamental cause of why MPC fails under stringent steering constraints is still unknown. 

### Expected behavior

The MPC can calculate target steering with a small steering rate limit

### Actual behavior

The MPC fails to calculate target steering with a small steering rate limit

### Steps to reproduce

Please see the details in the previously reported issue https://github.com/autowarefoundation/autoware.universe/issues/4015.

The MPC parameter should have a small steering rate limit. In the previous issue, the values are as follows.

```
    steer_rate_lim_dps_list_by_curvature: [10.0, 20.0, 30.0]
    curvature_list_for_steer_rate_lim: [0.001, 0.002, 0.01]   
    steer_rate_lim_dps_list_by_velocity: [40.0, 30.0, 20.0] 
    velocity_list_for_steer_rate_lim: [10.0, 15.0, 20.0]  
```

kev278, 2023-12-30:
Hello @TakaHoribe 
I am planning to work on this problem. Looks like you implemented the whole mpc_lateral_controller, and I was wondering if you had some reference paper or other relevant source to view the literature and the approach that you have implemented?

maxime-clem, 2024-01-18:
@kev278 Thank you for checking this issue.
The [documentation of the MPC lateral controller](https://autowarefoundation.github.io/autoware.universe/main/control/mpc_lateral_controller/) contains the following reference:
> Jarrod M. Snider, "Automatic Steering Methods for Autonomous Automobile Path Tracking", Robotics Institute, Carnegie Mellon University, February 2009.

There is also a page explaining the MPC algorithm in details: https://autowarefoundation.github.io/autoware.universe/main/control/mpc_lateral_controller/model_predictive_control_algorithm/