autowarefoundation/autoware_universe #4035: No longitudinal velocity published from trajectory follower

yaukaizhi, 2023-06-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

No longitudinal speed output from trajectory follower, only lateral. Even though scenario planning outputs both.

**Velocities from /planning/scenario_planning/trajectory**
[image]

**Velocities from /control/trajectory_follower/control_cmd**
[image]

### Expected behavior

Both speeds output from trajectory follower

### Actual behavior

No longitudinal speed output from trajectory follower, only lateral. Even though scenario planning outputs both.

### Steps to reproduce

Run autoware.launch.xml with feedback from /velocity_status and /steering_status

yaukaizhi, 2023-06-21:
[launch.log](https://github.com/autowarefoundation/autoware.universe/files/11817343/launch.log)
Here you go!

maxime-clem, 2023-06-21:
```
1687348576.9311931 [component_container_mt-62] [33m[WARN 1687348576.930780846] [planning.scenario_planning.lane_driving.motion_planning.surround_obstacle_checker]: do not start because there is obstacle near the ego vehicle. (onTimer() at /home/kai/autoware/src/universe/autoware.universe/planning/surround_obstacle_checker/src/node.cpp:244)[0m
```
There are some lidar points very close to the vehicle position which prevent the vehicle to move.

```
1687348585.1372814 [component_container-66] [33m[WARN 1687348585.130116987] [control.trajectory_follower.controller_node_exe]: Too large position error: 107.759646m > 5.000000m (warn_throttle() at /home/kai/autoware/src/universe/autoware.universe/control/mpc_lateral_controller/include/mpc_lateral_controller/mpc.hpp:394)[0m
```
The vehicle pose is too far from the trajectory.

If you are replaying a bag, you may be having issues caused by conflicting topics that are published from 2 sources: the bag and the currently running Autoware.

yaukaizhi, 2023-06-22:
The issue seems to be because enable_keep_stopped_until_steer_convergence==true. It works when I switch it to false. Thanks for the tip on using the log file!

May I know why you decided to make steer convergence on by default? Wouldn't that affect 90 degree turns?

closed 2023-06-22 by yaukaizhi