autowarefoundation/autoware_universe #4062: Localization Validation Error Reference (time difference too large)

yaukaizhi, 2023-06-23:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Received the following error:
[localization.pose_estimator.ndt_scan_matcher]: Validation error. The reference time is 2071.972686[sec], but the target time is 1687506272.258270[sec]. The difference is 1687504200.285584[sec] (the tolerance is 1.000000[sec]). (validate_time_stamp_difference() at /home/kai/autoware/src/universe/autoware.universe/localization/ndt_scan_matcher/src/pose_array_interpolator.cpp:85)

Using physical lidar, works with rosbag and use_sim_time==true, but once use_sim_time==false then receive this error. What does this error mean?

### Expected behavior

Smooth launching without validation error

### Actual behavior

As shown above

### Steps to reproduce

Physical lidar testing with autoware.launch.xml

### Additional context

log file for reference
[launch.log](https://github.com/autowarefoundation/autoware.universe/files/11846105/launch.log)

kminoda, 2023-06-23:
It's from here: https://github.com/autowarefoundation/autoware.universe/blob/25f9d6a03adf91ed2312741b9acdf937560245a9/localization/ndt_scan_matcher/src/pose_array_interpolator.cpp#L78-L92
Which probably is called here: https://github.com/autowarefoundation/autoware.universe/blob/e70df1239c67d8ae86a7a8bf8d6c83bc20e7ece7/localization/ndt_scan_matcher/src/ndt_scan_matcher_core.cpp#L362C25-L364

Since ndt_scan_matcher needs an initial guess pose for optimization, it uses the EKF outputs. But since we need to use a proper EKF outputs with a close timestamp with the received LiDAR timestamp, we add the above timestamp validation. Without it, Autoware may use old EKF pose output and NDT ends up in non-optimal poses.

yaukaizhi, 2023-06-23:
So what would cause the timestamps to have such a large difference?

kminoda, 2023-06-23:
use_sim_time==false probably is the direct cause

yaukaizhi, 2023-06-23:
but if I set use_sim_time=true, the LIDAR pointcloud becomes sluggish as if it's in slowmo. 
 
What might cause this?

kminoda, 2023-06-23:
Now I got the issue :+1: I think it's better to change the title as the current one is a bit misleading. It seems it's not about specific package's implementation but rather an issue as an whole system.

Could you elaborate more on how it's sluggish? You mean on RViz? Could you share any videos that captures the issue here? Please also make it clear when exactly that issue happens. Is it when you are using `logging_simulator` with your recorded rosbag? Or using `autoware.launch.xml` with your onboard PC on your real robot or something?

yaukaizhi, 2023-06-23:
This happens when using logging_simulator or autoware.launch.xml, use_sim_time=true. It does not occur when ros2 bag play <bag name> --clock. 

https://github.com/autowarefoundation/autoware.universe/assets/112715209/cd5c6c35-c051-458f-b026-a52e649a83aa

kminoda, 2023-06-24:
You should not use autoware.launch.xml with use_sim_time=true as long as you are not providing /clock. So your rosbag should also contain /clock as well

yaukaizhi, 2023-06-24:
The issue has been solved by ensuring the LIDAR driver syncs with system time on my laptop. Thanks!

closed 2023-06-24 by yaukaizhi