autowarefoundation/autoware_universe #4078: refactor: similar combine lanelets functions

kosuke55, 2023-06-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

There are similar functions, better to replace with a common one.
- mission_planner https://github.com/autowarefoundation/autoware.universe/blob/main/planning/mission_planner/src/lanelet2_plugins/utility_functions.cpp#L57 
- lanelet2_extension https://github.com/autowarefoundation/autoware_common/blob/main/tmp/lanelet2_extension/lib/utilities.cpp#L232

### Purpose

refactor 

### Possible approaches

replace with common

### Definition of done

replace with common

mehmetdogru, 2023-08-17:
@kosuke55 san is the issue resolved?

kosuke55, 2023-08-17:
@mehmetdogru 
Not yet, the corresponding function in mission_planner needs to be replaced :pray:

kosuke55, 2023-11-09:
solved by
https://github.com/autowarefoundation/autoware_common/pull/193
https://github.com/autowarefoundation/autoware.universe/pull/5535

closed 2023-11-09 by kosuke55