autowarefoundation/autoware_universe #4143: Planning module fails to generate a proper path to the goal right behind the car

Kim-mins, 2023-07-03:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Running environment: [docker image](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/102247121?tag=humble-20230616-prebuilt-cuda-amd64)

While I run Autoware on Carla Map Town01 and set the destination to the right behind of the ego vehicle, Autoware fails to drive as usual. [[video](https://drive.google.com/file/d/1hD6eVF-VH4JUzesPy8Zbo6R_XMw56PNI/view?usp=sharing)] [[rosbag file](https://drive.google.com/drive/folders/16m0IFk7pKg9WczSrWNafBiZ2qnawKyX0?usp=sharing)]
In the video, I changed the goal again and again, and I finally found that the planning module works well when I set the goal point behind the car but not on the same lane(the lane defined from openstreetmap file)

### Expected behavior

I expected the car to drive, even when I selected the point right behind him as a goal.

### Actual behavior

It seems like there are some misbehaviors on the planning module and the car does not move at all.

### Steps to reproduce

1. Run Autoware on docker container [[image](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/102247121?tag=humble-20230616-prebuilt-cuda-amd64)] [[map](https://drive.google.com/drive/folders/1JbsUAWIwNWHe5cBQvKGpRGMd1esLcAcj?usp=sharing)]
2. Set the goal right behind the car.

### Versions

I build and run on the docker container with the image [here](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/102247121?tag=humble-20230616-prebuilt-cuda-amd64) (same with the image link right above)

### Possible causes

Sorry. I'm novice to Autoware, so I cannot localize the possible causes well.
But it seems like the problem is on planning module, since I've observed that the planned trajectory starts from the front of the ego vehicle, but it does not in this case.

kosuke55, 2023-07-10:
thanks for the report.
as you found currently the goal behind ego in the same lene is not supported and there is a similar issue https://github.com/autowarefoundation/autoware.universe/issues/2354

Kim-mins, 2023-07-10:
Oh, I should have searched more.. Sorry for bothering you. I didn't noticed there was the same issue.

closed 2023-07-12 by Kim-mins