autowarefoundation/autoware_universe #4153: Intersection module dies when object detected at some turns

ismetatabay, 2023-07-03:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

As in the video, as the vehicle approaches the intersection, the behavior_planning_component also dies because the intersection module dies. 

https://github.com/autowarefoundation/autoware.universe/assets/56237550/9866b042-f0c4-4dfa-b67a-ec2ca6e954d6

### Expected behavior

Intersection Module and Behavior Planning Container works well without dying.

### Actual behavior

Intersection Module dies, so Behavor Pllanning Container dies.

### Steps to reproduce

- Use this lanelet2 map
[lanelet2_map.txt](https://github.com/autowarefoundation/autoware.universe/files/11937249/lanelet2_map.txt)
- Follow same steps as video in the description

### Versions

- OS: Ubuntu 22.04
- ROS: ROS 2 Humble 
- Autoware: Latest version of Autoware.Universe

### Possible causes

The reason why the Intersection module dies is because the calculated ["start_arc_length"](https://github.com/autowarefoundation/autoware.universe/blob/56b9ed8587098e95eec243dc41df0840bfadcb65/planning/behavior_velocity_intersection_module/src/scene_intersection.cpp#L996C22-L996C38) and ["end_arc_length"](https://github.com/autowarefoundation/autoware.universe/blob/56b9ed8587098e95eec243dc41df0840bfadcb65/planning/behavior_velocity_intersection_module/src/scene_intersection.cpp#L999) values are bigger than the combined ```lanelet::ConstLanelets``` in the ```ego_lane_with_next_lane```. 

So ["trimmed_ego_polygon"](https://github.com/autowarefoundation/autoware.universe/blob/56b9ed8587098e95eec243dc41df0840bfadcb65/planning/behavior_velocity_intersection_module/src/scene_intersection.cpp#L1002) becomes empty polygon and this line causes dying of intersection modules. 
https://github.com/autowarefoundation/autoware.universe/blob/56b9ed8587098e95eec243dc41df0840bfadcb65/planning/behavior_velocity_intersection_module/src/scene_intersection.cpp#L1010

ismetatabay, 2023-07-03:
@soblin Hi Sobue, have a nice week. There is 2 approach which I think:

- First, we can check the `trimmed_ego_polygon` is empty or not, so if estimated `start_arc_length` and `end_arc_length` are outside (bigger) than the combined `ego_lane_with_next_lane` lanelets length, maybe we can say there is no collusion, but  this approach can be dangerous for edge cases.
- Second, we can add new `getEgoLaneWithNextLane` method which takes arc lengths as parameter, and add lanelets to `ego_lane_with_next_lane` until total lengths of added lanelets is bigger than arc lengths. 

What do you think?

satoshi-ota, 2023-07-13:
Hi @soblin Could you answer the question?

soblin, 2023-07-13:
I'm sorry I was confusing with this PR which is already merged.

https://github.com/autowarefoundation/autoware.universe/pull/4191

I’ll take a look at this PR
https://github.com/autowarefoundation/autoware.universe/pull/4177

beyzanurkaya, 2023-07-13:
@soblin The intersection module was dying for 2 different reasons. I realized the second reason later because a PR that merged later was the reason, and for this reason, I created 2 PR.

closed 2023-07-20 by beyzanurkaya