autowarefoundation/autoware_universe #4178: Avoidance module dies in U turns with spesific conditions

ismetatabay, 2023-07-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

If target objects included at different lanelets on reference path with opposite direction as in the image, avoidance modules dies because at this error throwing.

### Expected behavior

Avoidance module works well. The module must create separate avoid lines for different section of same object. 

### Actual behavior

Avoidance module creates huge shift and the modules dies.

### Steps to reproduce

This problem can be reproduced as this video on planning simulator:

https://github.com/autowarefoundation/autoware.universe/assets/56237550/4e0b47b2-0bae-45c5-844a-2cb892bef823

### Versions

- OS: Ubuntu 22.04
- ROS: ROS 2 Humble
- Autoware: Latest version of Autoware.Universe

### Possible causes

This lines throws error and module dies:

https://github.com/autowarefoundation/autoware.universe/blob/c9280db0f945217d9f95e49d2f76ee4adab1fdac/planning/behavior_path_planner/include/behavior_path_planner/utils/avoidance/helper.hpp#L60-L62

Because, at some U turns cases, these situation can be occur:
[image]

takayuki5168, 2023-07-06:
cc @satoshi-ota

mehmetdogru, 2023-07-12:
@satoshi-ota - san, do you think you will be working on this task any time soon? It is kind of an urgency on our end and we can prioritize this task and start working on it if it is ok.

satoshi-ota, 2023-07-12:
Hi @mehmetdogru @ismetatabay Thanks for your bug reports.

> do you think you will be working on this task any time soon?

Yeah, I can work on this with high priority :+1: But I would like to get the map (I want only lanelet2_map.osm). Is it possible to share the file with me?

satoshi-ota, 2023-07-13:
> Hi @mehmetdogru @ismetatabay Thanks for your bug reports.
> 
> > do you think you will be working on this task any time soon?
> 
> Yeah, I can work on this with high priority +1 But I would like to get the map (I want only lanelet2_map.osm). Is it possible to share the file with me?

@mehmetdogru @ismetatabay I could reproduce the issue in tokyo map. So, I'll start debug right away.

ismetatabay, 2023-07-13:
@satoshi-ota Thanks a lot for your PR, it solves dying problem :+1: , avoidance module not dies anymore. Moreover as we talked with @mehmetdogru, at this case, we think there should be more than one avoidance line.

The image shows that there is one huge shifting line for object (the shifting line length is approx ~150m): 
[image]

Maybe we can create different avoidance shifting lines for same object which included in different lanelets. This situation can be topic for another issue and another PR, what dou you think?

satoshi-ota, 2023-07-13:
@ismetatabay @mehmetdogru 

> Thanks a lot for your PR, it solves dying problem +1 
Great. Thanks for your checking.

> Maybe we can create different avoidance shifting lines for same object which included in different lanelets. This situation can be topic for another issue and another PR, what dou you think?
Yeah, I agree with you. Please open another issue, and maybe I can work on your proposak next week.

satoshi-ota, 2023-07-13:
@ismetatabay @mehmetdogru Unfortunately, I don't have map that contains such a sharp U-turn road. So, I would be happly if you could share the map with me.

ismetatabay, 2023-07-13:
@satoshi-ota Of course, here is the map:
[lanelet2_map.txt](https://github.com/autowarefoundation/autoware.universe/files/12037186/lanelet2_map.txt)

The intersection can be shown as in the video which included in this Issue description:
- https://github.com/autowarefoundation/autoware.universe/issues/4153