autowarefoundation/autoware_universe #4188: Adding option for give median position in gnss_poser

meliketanrikulu, 2023-07-06:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Gnss_poser give position as median position. İts good option if gnss accuracy is low. But we want to use directly gnss position. So maybe this option can be added as a parameter.

### Purpose

It may be suitable for those who want to get from gnss poser according to the location coming directly from gnss. In the future, it will be more appropriate to use the real gnss location of gnss_poser to take localization output with gnss and include it in the ekf.

### Possible approaches

In this line gnss_posers output position is median position. Can be add if structure that associated with a parameter in the parameter file.

### Definition of done

Being able to use gnss poses directly and using median position parameters can be selected from the config file

YamatoAndo, 2023-07-07:
@meliketanrikulu 
It is indeed supposed to take the median, but since [the number of elements defaults to 1](https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/gnss_poser/src/gnss_poser_core.cpp#L48-L49).
So it is the same as the raw value, isn't it?

YamatoAndo, 2023-07-07:
Be that as it may, I think it would be nice to have a parameter that explicitly states whether or not the median should be used.

yvzksgl, 2023-07-12:
I have a couple of questions. 

- What is the reason of publishing median value? As far as i understand it works same with publishing coming data immediately, if ```buffer_epoch``` setted as 1. If ```buffer_epoch``` increased what is the effect of publishing median value?
- I have an idea that provides 3 options as
       1. Publish coming data immediately (ıt will set buffer_epoch as1.).
       2. Publish median of the buffer (buffer_epoch setted by user.).
       3. Publish average value of the poses in the buffer (buffer_epoch setted by user.).

YamatoAndo, 2023-07-13:
@yvzksgl 
The reason for using the median is to remove outliers. In the past, there were occasional large jumps in GNSS positioning values, so we used the median to prevent that. However, with the recent significant improvement in positioning accuracy and the increasing prevalence of high-speed autonomous driving, we no longer rely on the use of the median.

Therefore, we have made it the default to publish coming data immediately, without using the median. However, I believe it would be beneficial to provide options for using the median or publishing the median values under a different topic name.

yvzksgl, 2023-07-19:
In my point of view, it would be better let the user choose what to be published in the existing topic instead of publishing separate topics. If it sounds okay i will modify the node as i stated before, keep the topic same and what to publish will be selected with parameters:

>       1. Publish coming data immediately (ıt will set buffer_epoch as1.).
>       2. Publish median of the buffer (buffer_epoch setted by user.).
>       3. Publish average value of the poses in the buffer (buffer_epoch setted by user.).

closed 2023-07-24 by YamatoAndo