autowarefoundation/autoware_universe #4267: [pid_longitudinal_controller]velocity error compute error when hill start

ahuazuipiaoliang, 2023-07-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

when the vehicle is on a slope and go uphill, the vehicle may going backwards when it stops, because of the error of slope estimation or/and pedal calibration. In this case,  the `velocity_error` of `PID controller` should increase because target velocity is position or 0, and the current velocity is negative. 

### Expected behavior

The expected ` velocity_error = target_vel - current_vel`.

### Actual behavior

[image]
The actual `velocity_error = abs(target_vel) - abs(current_vel)`
For example, when the `target_vel` is 1.0, `current_vel` is -1.0, the expected `velocity_error` should be 2.0, but the actual `velocity_error` is 0.0.

### Steps to reproduce

(pass)

### Versions

- OS: Ubuntu 22.04
- ROS: Humble
- autoware_universe: 56f19436cd7f6c09e0e8d00065f8ff39dbb633eb

ahuazuipiaoliang, 2023-07-13:
@TakaHoribe I think you are the chair of Planning&Control WG, please have a look at this issue.

ahuazuipiaoliang, 2023-07-18:
@BonoloAWF my colleague is working on it and will create a PR

TakaHoribe, 2023-07-18:
@ahuazuipiaoliang Thank you for the issue report. This is definitely an issue and your proposal is totally make sense. Looking forward to your PR. Thank you!

ahuazuipiaoliang, 2023-09-22:
@brkay54 seems resolved,

closed 2023-09-22 by ahuazuipiaoliang

takayuki5168, 2023-09-23:
@ahuazuipiaoliang @brkay54 @lixianqiang
Was the PR really merged? I cannot find.

takayuki5168, 2023-09-23:
@ahuazuipiaoliang @brkay54 @lixianqiang cc @TakaHoribe 
I created a PR.
The change is a bit more complex than just using `velocity_error = target_vel - current_vel`.
https://github.com/autowarefoundation/autoware.universe/pull/5094