autowarefoundation/autoware_universe #4276: Refactoring the perception packages parameter definitions

ismetatabay, 2023-07-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

At the current design of Autoware.Universe architecture, most parameters migrates along launch files. (with arguments)

For example, if we want to modify gnss topic names for gnss poser, we must change parameter values in sample_sensor_kit_launch -> gnss.launch.xml:
- https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/5a66ef0e14e21d5ce13eff16ca63cfcf0cfb7798/sample_sensor_kit_launch/launch/gnss.launch.xml#L9-L12

### Purpose

Autoware users will not need modify the Autoware.Universe or sensor_kit_launch repository, they can change necessary parameters on autoware_launch package.

### Possible approaches

Convert the launch file arguments to ROS 2 param.yaml and create necessary param files to autoware_launch

### Definition of done

Definition of done (WIP)

- Prepare the necessary updates on sensing nodes
  - gnss_poser:
    - read topic names as a parameter from .yaml file

mitsudome-r, 2023-07-14:
@ismetatabay 
Thank you for posting the issue.

I don't mind extracting the parameters hard coded in the launch file as a parameter file, but I would like to discuss a little about where we put them. Since gnss.launch.xml exists in sample_sensor_launch package, I think it is simpler to place it under sample_sensor_launch's config directory.

> Autoware users will not need modify the Autoware.Universe or sensor_kit_launch repository, they can change necessary parameters on autoware_launch package.

Since the sensor configuration is user dependent, our assumption is that the users will fork and modify the sensor_kit_launch for their vehicle anyways. (e.g. people might have no gnss or multiple gnss) Therefore, I think it is easier for such users to modify the parameters in the same package/repository as they customize the launch file.

ismetatabay, 2023-07-14:
@mitsudome-r Thank you for your valuable comment, you are right. I think we shouldn't migrate sensing parameters to autoware_launch. I will create another PR for adding [missing gnss_poser parameter](https://github.com/autowarefoundation/autoware.universe/blob/ef3f0cb1a1b061f6fbb4c9fd5368b3e7617f2139/sensing/gnss_poser/launch/gnss_poser.launch.xml#L29-L39)  to [sample_sensor_kit_launch](https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/5a66ef0e14e21d5ce13eff16ca63cfcf0cfb7798/sample_sensor_kit_launch/launch/gnss.launch.xml#L29-L39). :+1:

closed 2023-07-14 by ismetatabay