autowarefoundation/autoware_universe #4279: fix: dead github workflows | spell-check-differential

HansRobo, 2023-07-14:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Currently, spell-check-differential is one of the CI executed by pull requests as a trigger.
However, since there are too many cspell errors, the error is almost ignored when pull-request is merged.

So, it is desirable to reduce typos and revive spell-check-differential workflow!
if we archieve zero cspell errors, we have the option to set the CI to "required".

### Purpose

To revive the dead spell-check-differential workflow

### Possible approaches

We can...
- fix the typos
- add the words to [dictionary](https://github.com/tier4/autoware-spell-check-dict)
- add ignore comment inline

We can track the number of entire cspell errors by [spell-check-all workflow](https://github.com/autowarefoundation/autoware.universe/pull/4272) 

### Definition of done

Pass the [spell-check-all workflow](https://github.com/autowarefoundation/autoware.universe/pull/4272)

HansRobo, 2023-07-18:
https://github.com/autowarefoundation/autoware.universe/pull/4280

HansRobo, 2023-07-18:
Errors: [730](https://github.com/autowarefoundation/autoware.universe/actions/runs/5583469469)
main@2023/07/18 1:55P.M. JST

HansRobo, 2023-07-18:
We introduced `spell-check-partial` workflow to make spell-check "required" step by step.
https://github.com/autowarefoundation/autoware.universe/pull/4320

HansRobo, 2023-07-19:
@kminoda is trying to test the `spell-check-partial` is works fine!

https://github.com/autowarefoundation/autoware.universe/pull/4321
https://github.com/autowarefoundation/autoware.universe/pull/4329
https://github.com/autowarefoundation/autoware.universe/pull/4331

and he also prepare for actual operation of `spell-check-partial`!
https://github.com/autowarefoundation/autoware.universe/pull/4327

HansRobo, 2023-07-19:
Errors: [668](https://github.com/autowarefoundation/autoware.universe/actions/runs/5594306358)
main@2023/07/19 11:31 A.M.

This pull request works great for the reduction of errors this time!
https://github.com/autowarefoundation/autoware.universe/pull/4281

HansRobo, 2023-07-19:
WIP for next error observation
https://github.com/autowarefoundation/autoware.universe/pull/4332

kminoda, 2023-07-19:
Memo
spell-check-partial has been made required

HansRobo, 2023-07-19:
Errors: [667](https://github.com/autowarefoundation/autoware.universe/actions/runs/5599696834)
main@2023/07/19 22:22

https://github.com/autowarefoundation/autoware.universe/pull/4332

HansRobo, 2023-07-21:
Errors: [695](https://github.com/autowarefoundation/autoware.universe/actions/runs/5624486640)
main@2023/07/22 1:30 JST

HansRobo, 2023-07-22:
Errors: [689](https://github.com/autowarefoundation/autoware.universe/actions/runs/5627969082)
main@2023/07/22 10:25 JST

HansRobo, 2023-07-24:
Errors: [680](https://github.com/autowarefoundation/autoware.universe/actions/runs/5641241035)
main@2023/0724 14:39 JST

HansRobo, 2023-07-24:
`spell-check-partial` is enabled for `localization`!
https://github.com/autowarefoundation/autoware.universe/pull/4377

HansRobo, 2023-07-26:
`spell-check-partial` is enabled for `docs`, `simulator`, `tools`, and `vehicle`!!
https://github.com/autowarefoundation/autoware.universe/pull/4327

HansRobo, 2023-07-26:
Errors: [526](https://github.com/autowarefoundation/autoware.universe/actions/runs/5665090933)
main@2023/0726 14:42 JST

https://github.com/autowarefoundation/autoware.universe/pull/4289

HansRobo, 2023-07-27:
Errors: [517](https://github.com/autowarefoundation/autoware.universe/actions/runs/5675767770)
main@2023/07/27 11:15 JST

HansRobo, 2023-07-27:
`spell-check-partial` is enabled for `common`
https://github.com/autowarefoundation/autoware.universe/pull/4397

HansRobo, 2023-07-28:
Errors: [501](https://github.com/autowarefoundation/autoware.universe/actions/runs/5687779147)
main@2023/07/28 12:07 JST

https://github.com/autowarefoundation/autoware.universe/pull/4403
https://github.com/autowarefoundation/autoware.universe/pull/4404
https://github.com/autowarefoundation/autoware.universe/pull/4406

HansRobo, 2023-07-31:
`spell-check-partial` is enabled for `system`
https://github.com/autowarefoundation/autoware.universe/pull/4460

HansRobo, 2023-07-31:
`spell-check-partial` will be enabled for `launch` and `evaluator`
https://github.com/autowarefoundation/autoware.universe/pull/4462

HansRobo, 2023-08-03:
Errors: [476](https://github.com/autowarefoundation/autoware.universe/actions/runs/5733132938)
main@2023/08/02 9:13 JST

HansRobo, 2023-08-21:
Errors: [477](https://github.com/autowarefoundation/autoware.universe/issues/4279)
main@2023/08/07 11:45 JST

HansRobo, 2023-08-21:
Errors: [476](https://github.com/autowarefoundation/autoware.universe/actions/runs/5806829839)
main@2023/08/09 17:41 JST

HansRobo, 2023-08-21:
Errors: [476](https://github.com/autowarefoundation/autoware.universe/actions/runs/5815783444)
main@2023/08/10 10:16 JST

HansRobo, 2023-08-21:
Errors: [434](https://github.com/autowarefoundation/autoware.universe/issues/4279)
main@2023/08/18 14:56 JST

#4613

HansRobo, 2023-08-21:
Errors: [430](https://github.com/autowarefoundation/autoware.universe/issues/4279)
main@2023/08/21 14:04 JST

kminoda, 2023-08-21:
Spell-check errors for each components (and some major packages) as of 2023/08/21
- perception: 292
  - perception/bytetrack: 233 (already ignored in spell-check-partial)
  - perception/radar_object_tracker: 28
- planning: 106
  - planning/sampling_based_planner: 39
- sensing: 32

kminoda, 2023-08-22:
Errors: [372](https://github.com/autowarefoundation/autoware.universe/actions/runs/5933390984/job/16088753404#step:11:804)
main@2023/08/22 10:48 JST

HansRobo, 2023-08-23:
Errors: [328](https://github.com/autowarefoundation/autoware.universe/actions/runs/5945688652)
main@2023/08/23 9:57 JST

https://github.com/tier4/autoware-spell-check-dict/pull/616

HansRobo, 2023-08-24:
Errors: [305](https://github.com/autowarefoundation/autoware.universe/actions/runs/5961576072)
main@2023/08/24 17:52 JST

https://github.com/autowarefoundation/autoware.universe/pull/4716

mitsudome-r, 2026-03-24:
I'm closing the issue that was inactive for a long time. Feel free to reopen if the issue is still relevant.