autowarefoundation/autoware_universe #4349: Shotage of the disk space of GitHub-hosted runners

shmpwk, 2023-07-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Usually, `build-and-test-differential.yaml` CI in `autoware.universe` fails because of the shotage of the disk space.
Without passing the CI, PRs cannot be merged.

For example, this CI is failed, saying "no space left on device".
https://github.com/autowarefoundation/autoware.universe/actions/runs/5612516522 

### Purpose

Avoiding shotage of the disk space

### Possible approaches

- Use self-hosted runners

- Add billing of GitHub-hosted runners to increase the space

- Lighten up Autoware

- Adjust failing CI  

### Definition of done

No CI failed due to the shotage of the disk space.

xmfcx, 2023-07-21:
To fix this temporarily, we've assigned self hosted x64 runner machines to do this job.
- https://github.com/autowarefoundation/autoware.universe/pull/4364

There is only a single machine so far, there is going to be some queue.

For permanent fix, we need to focus on https://github.com/orgs/autowarefoundation/discussions/3469 to reduce the dev-env-docker file.

Or we should add more self hosted x64 machines to share the work.

shmpwk, 2023-07-22:
Thank you for sharing self hosted machines! 
I will close this task since it avoids the shotage problem for a while.
For long term fix, we also discuss it ourselves.

closed 2023-07-22 by shmpwk