autowarefoundation/autoware_universe #4350: Ego vehicle cannot turn right even though there are no obstacles

Kim-mins, 2023-07-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team, I'm currently running Autoware on Carla, and I found that the ego vehicle cannot turn right even though there are no obstacles.

You can refer to the video [here](https://drive.google.com/file/d/1m8gQJC_LIVwr0EEQEj6tPY6VRAgQ7W5_/view?usp=sharing), and you can see that there are no obstacles, but the vehicle failed to turn right.

The map is [carla Town01](https://carla.readthedocs.io/en/0.9.12/img/Town01.jpg), and I've observed that this situation occurs on every outer-most corners of the map.

### Expected behavior

The ego vehicle successfully turns right, since there's no interference.

### Actual behavior

But it can't.

### Steps to reproduce

[Here](https://drive.google.com/drive/folders/1DPIwh4Q2kPVM26WaiG1Nq1lMWSCBmfpA?usp=sharing)'s the ros2 bag file.

Or if you want to reproduce it yourself, 
1. please download the following map files: [[lanelet2](https://drive.google.com/file/d/1vm9SvalJe7Bc9sh_8jK-0cPulVAov5uD/view?usp=drive_link)] [[pcd](https://drive.google.com/file/d/1MvJlfjkw3LWFUs5hdE_KQ2_CQTusU8UN/view?usp=drive_link)]
2. Run Autoware and spawn the vehicle at the point near to the corner and set a goal to pass the corner, just as the video above.

 
Please note that, this issue sometimes does not happen, but I think you can reproduce it in high probability.

### Versions

I ran the experiment on the container from the image [here](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/70625594?tag=humble-20230215-prebuilt-cuda-amd64).
Carla version: 0.9.13

### Possible causes

To my knowledge, the cause is from `obstacle_stop_planner` since it manages a stopping plan while driving.

maxime-clem, 2023-07-21:
The vehicles stops because, based on the obstacle pointcloud, there obstacles are on the path.
The default pointcloud processing is maybe not suited to the CARLA simulation and additional filtering may be necessary to solve your issue.

Kim-mins, 2023-07-21:
Thank you @maxime-clem! I just noticed the pointcloud information. 
Maybe the point cloud map I'm using now has some errors. Thank you for answering.

maxime-clem, 2023-07-21:
As far as I know the pointcloud map is only used for localization.
The obstacle pointcloud is obtained from the lidar sensor after ground removal (https://autowarefoundation.github.io/autoware.universe/main/perception/ground_segmentation/). In your case, it looks like the ground is not correctly filtered.

Kim-mins, 2023-07-21:
Thank you for the follow up comment @maxime-clem. I think I got your point.
In my understanding, maybe you mean there's a misbehavior on the ground removal process of perception module when Autoware runs on Carla simulator.

By the way, when I run Autoware on Carla, this issue interferes my simulation too frequently and it stops me from running Autoware normally. 
How's your plan? Can this issue be fixed soon, or in low priority?

maxime-clem, 2023-07-21:
I personally do not use Carla so I cannot work on this issue. Since it is not officially supported by Autoware there are not many people who can help. I suggest trying to tune the parameters of the ground filter by yourself.
Otherwise, you can try tuning the [`obstacle_stop_planner`](https://autowarefoundation.github.io/autoware.universe/main/planning/obstacle_stop_planner/), for example by setting `enable_z_axis_obstacle_filtering` to `true`.

Kim-mins, 2023-07-23:
Thank you for answering @maxime-clem! I'll try your suggestion soon.
Then, should I close this issue or not? This issue seems not urgent and only occurs in carla. 
How do you think? I can follow your issue management policy.

maxime-clem, 2023-07-24:
It is okay to keep the issue open and I updated the title of the issue to make it easier to track.

Kim-mins, 2023-08-07:
Hi @maxime-clem!
I have a good news that, this issue is gone on the recent version of Autoware(2023/07/15).
While I was testing a recent version of Autoware for [another issue that you answered](https://github.com/autowarefoundation/autoware.universe/issues/4505#issue-1834151069), I could find that the problem is resolved.
Maybe the issue can be closed. Thank you so much!

closed 2023-08-08 by Kim-mins