autowarefoundation/autoware_universe #4359: Reduce binary size on the CI

soblin, 2023-07-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

By passing `-Os` option to the compiler, the object file is optimized in terms of the output size. Then we can also pass `-Wl, s` option to the linker to remove symbol table from the executble. I compiled one file `scene_intersection.cpp` in `behavior_velocity_intersection_module` and checked how much the size is reduced.

- By default build command: 46.6MB
- with `-Os` option: 32.6MB

At least for this file the size is reduced by 30%. The sum of object file (`*.o`, `*.so`, excluding the executable) size under `build/` directory was 2.268GB on my machine according to the following command.

```
find ./ -type f -name "*.so" -o -name "*.o" | xargs du -c
```

I guess we can expect to save the disk size by `2.268 * 0.3 = 0.68` GB.

### Purpose

- avoid `No space left on disk`

### Possible approaches

- Apply `g++ -c -Os -Wl, s` option

### Definition of done

- Change the compile option only on the CI

BonoloAWF, 2023-07-21:
cc: @xmfcx @mitsudome-r

xmfcx, 2023-07-21:
### With flags
Run following:
`colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS=1 -DCMAKE_C_FLAGS="-Os -Wl,-s" -DCMAKE_CXX_FLAGS="-Os -Wl,-s"` on a recent branch of autoware.

```bash
autoware_2$ du -sh build install
3,8G	build
130M	install
```

### Without flags
And run following:
`colcon build --symlink-install  --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS=1 `

```bash
autoware$ du -sh build install
4,6G	build
138M	install
```

### Comparison
These flags might be overridden by things in [autoware_package.cmake](https://github.com/autowarefoundation/autoware_common/blob/main/autoware_cmake/cmake/autoware_package.cmake) (didn't have time to check thoroughly) but worst case scenario, we can add the flags here.

But even without making changes in the repository, it managed 17.39% disk space usage reduction.

### Additional info
Will also try:
`-flto : This enables Link Time Optimization (LTO). It allows the compiler to optimize at application scope rather than file scope. This flag can lead to better optimization, but increases compile time.` to see if it reduces the disk space used (probably not).

Also my test might be a bit flawed too, i think these 2 folders of autoware weren't identical (but close). Will retest it on Monday.

closed 2023-09-21 by soblin