autowarefoundation/autoware_universe #4386: Localization sometimes misbehaves while turning

Kim-mins, 2023-07-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware on Carla, and I found that the localization module sometimes does not work well while turning.

[video1]() here shows that localization works well at first, but it gets twisted while turning, and it is not corrected.

[video2]() also shows that the localization has some problem while turning, but unlike video1, it stops at the corner(since there's a car in front of it). 
While stopping, it seems like correcting the localization error, and the ego vehicle eventually invades the drivable area and stucks.

Maybe this error is related to the issue [here](https://github.com/autowarefoundation/autoware.universe/issues/1426), but I report this issue since it is closed and expected to be fixed already..

### Expected behavior

Localization to be stable while turning

### Actual behavior

Localization is somewhat unstable while turning

### Steps to reproduce

Here are the ros2 bag files for [video1](https://drive.google.com/drive/folders/1GaE4ZjjOt_zNh9RCfl10Jij9lsOmb48A?usp=sharing) and [video2](https://drive.google.com/drive/folders/1yyjT-cVBHQofeq93m6g0yD7qARp5XzNQ?usp=sharing).

### Versions

I ran the experiment on the container from the image [here](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/70625594?tag=humble-20230215-prebuilt-cuda-amd64).
Carla version: 0.9.13

### Possible causes

I don't know about the deep details of Autoware, so I guess this issue is from localization module.

kminoda, 2023-07-25:
Hi, thank you for the report :+1:
Unfortunately, it is difficult to tell the exact cause without looking into the rosbag and knowing the context of this experiment. 

But let me provide you some starting point:
- Checking the output log of `ndt_scan_matcher` or visualizing some debugging scores and values from NDT and plotting them would be helpful for us to understand what's going on.
- Visualizing `/localization/pose_estimator/points_aligned` on RViz would be helpful to understand if it's truly a localization issue or not (see the below figure)

[image]

Kim-mins, 2023-07-25:
Thank you for the response @kminoda!!

I tried your **first suggestion** and here are the additional video:
- [For video1](https://drive.google.com/file/d/1rTUGMVzsYk8BuPpfddPmpQscxVxECnXe/view?usp=sharing)
- [For video2](https://drive.google.com/file/d/1p4eeBNR0I9KOfsyVn4-COxjPcz_fuVTt/view?usp=sharing)

For the **second suggestion**, I added the topic `/localization/pose_estimator/points_aligned` on RViz, but I could not find any difference of the visualization even though I clicked the check box of `Display` panel.. so I skipped it for now..

I'm a novice on Autoware, so maybe the provided information is not that meaningful to you.
So here are **the rosbag files for each video**, for better debugging. I hope these to be helpful:
- [For video1](https://drive.google.com/drive/folders/1GaE4ZjjOt_zNh9RCfl10Jij9lsOmb48A?usp=drive_link)
- [For video2](https://drive.google.com/drive/folders/1yyjT-cVBHQofeq93m6g0yD7qARp5XzNQ?usp=drive_link)

Thank you again, and please let me know if you need any additional data from me!

kminoda, 2023-07-25:
@Kim-mins Thank you for the quick response :+1: 

For the first suggestion (about the log of `ndt_scan_matcher`), one possible way is to see the terminal output of Autoware Launch to see `ndt_scan_matcher` outputs any warnings. You can simply check it with `rqt_console` or just launching Autoware with `| grep ndt_scan_matcher`.

For the second suggection (RViz visualization), your video is partially helpful, but I'd like to see a broader view of it. The basic idea behind this debugging is to check if the green squares (which is the direct output of `ndt_scan_matcher`) matches the pointcloud map. If it does matches, then NDT is working fine. If it does NOT match, then there is some issue in `ndt_scan_matcher`.

Kim-mins, 2023-07-25:
Thank you for the explanation for me @kminoda! I really appreciate to your help!
Maybe the distance between two arrows are not that important.. sorry for that.

Video and logs are here:

- For video1
    - [Additional video for video1](https://drive.google.com/file/d/1I8PbumMQt95rjaCKmDoaJoZKueGkk7bv/view?usp=sharing)
    <details>
    <summary>Log for video1 from Autoware by using `grep ndt_scan_matcher` </summary>
    
    ```
    [INFO] [ndt_scan_matcher-32]: process started with pid [11115]
    [aggregator_node-19] [WARN] [1690271527.901195636] [Aggregator]: No timestamp set for diagnostic message. Message names: ndt_scan_matcher
    [ndt_scan_matcher-32] [INFO] [1690271527.573486718] [localization.pose_estimator.ndt_scan_matcher]: points_queue_size: 1
    [ndt_scan_matcher-32] [INFO] [1690271527.573589156] [localization.pose_estimator.ndt_scan_matcher]: base_frame_id: base_link
    [ndt_scan_matcher-32] [INFO] [1690271527.573614365] [localization.pose_estimator.ndt_scan_matcher]: ndt_base_frame_id: ndt_base_link
    [ndt_scan_matcher-32] [INFO] [1690271527.573688729] [localization.pose_estimator.ndt_scan_matcher]: trans_epsilon: 0.010000, step_size: 0.100000, resolution: 2.000000, max_iterations: 30
    [ndt_scan_matcher-32] [INFO] [1690271528.589355355] [localization.pose_estimator.ndt_scan_matcher]: Waiting for pcd loader service. Check if the enable_differential_load in pointcloud_map_loader is set `true`.
    [zenoh-bridge-dds-70] [2023-07-25T07:52:08Z INFO  zenoh_plugin_dds] Route Zenoh->DDS (v1/rq/localization/pose_estimator/ndt_scan_matcher/get_parametersRequest -> rq/localization/pose_estimator/ndt_scan_matcher/get_parametersRequest): created with topic_type=rcl_interfaces::srv::dds_::GetParameters_Request_
    [zenoh-bridge-dds-70] [2023-07-25T07:52:08Z INFO  zenoh_plugin_dds] Route Zenoh->DDS (v1/rq/localization/pose_estimator/ndt_scan_matcher/get_parameter_typesRequest -> rq/localization/pose_estimator/ndt_scan_matcher/get_parameter_typesRequest): created with topic_type=rcl_interfaces::srv::dds_::GetParameterTypes_Request_
    [zenoh-bridge-dds-70] [2023-07-25T07:52:08Z INFO  zenoh_plugin_dds] Route Zenoh->DDS (v1/rq/localization/pose_estimator/ndt_scan_matcher/set_parametersRequest -> rq/localization/pose_estimator/ndt_scan_matcher/set_parametersRequest): created with topic_type=rcl_interfaces::srv::dds_::SetParameters_Request_
    [zenoh-bridge-dds-70] [2023-07-25T07:52:08Z INFO  zenoh_plugin_dds] Route Zenoh->DDS (v1/rq/localization/pose_estimator/ndt_scan_matcher/set_parameters_atomicallyRequest -> rq/localization/pose_estimator/ndt_scan_matcher/set_parameters_atomicallyRequest): created with topic_type=rcl_interfaces::srv::dds_::SetParametersAtomically_Request_
    [zenoh-bridge-dds-70] [2023-07-25T07:52:08Z INFO  zenoh_plugin_dds] Route Zenoh->DDS (v1/rq/localization/pose_estimator/ndt_scan_matcher/describe_parametersRequest -> rq/localization/pose_estimator/ndt_scan_matcher/describe_parametersRequest): created with topic_type=rcl_interfaces::srv::dds_::DescribeParameters_Request_
    [zenoh-bridge-dds-70] [2023-07-25T07:52:08Z INFO  zenoh_plugin_dds] Route Zenoh->DDS (v1/rq/localization/pose_estimator/ndt_scan_matcher/list_parametersRequest -> rq/localization/pose_estimator/ndt_scan_matcher/list_parametersRequest): created with topic_type=rcl_interfaces::srv::dds_::ListParameters_Request_
    [zenoh-bridge-dds-70] [2023-07-25T07:52:08Z INFO  zenoh_plugin_dds] Route DDS->Zenoh (rr/localization/pose_estimator/ndt_scan_matcher/get_parametersReply -> v1/rr/localization/pose_estimator/ndt_scan_matcher/get_parametersReply): created with topic_type=rcl_interfaces::srv::dds_::GetParameters_Response_
    [zenoh-bridge-dds-70] [2023-07-25T07:52:08Z INFO  zenoh_plugin_dds] Route DDS->Zenoh (rr/localization/pose_estimator/ndt_scan_matcher/get_parameter_typesReply -> v1/rr/localization/pose_estimator/ndt_scan_matcher/get_parameter_typesReply): created with topic_type=rcl_interfaces::srv::dds_::GetParameterTypes_Response_
    [zenoh-bridge-dds-70] [2023-07-25T07:52:08Z INFO  zenoh_plugin_dds] Route DDS->Zenoh (rr/localization/pose_estimator/ndt_scan_matcher/set_parametersReply -> v1/rr/localization/pose_estimator/ndt_scan_matcher/set_parametersReply): created with topic_type=rcl_interfaces::srv::dds_::SetParameters_Response_
    [zenoh-bridge-dds-70] [2023-07-25T07:52:08Z INFO  zenoh_plugin_dds] Route DDS->Zenoh (rr/localization/pose_estimator/ndt_scan_matcher/set_parameters_atomicallyReply -> v1/rr/localization/pose_estimator/ndt_scan_matcher/set_parameters_atomicallyReply): created with topic_type=rcl_interfaces::srv::dds_::SetParametersAtomically_Response_
    [zenoh-bridge-dds-70] [2023-07-25T07:52:08Z INFO  zenoh_plugin_dds] Route DDS->Zenoh (rr/localization/pose_estimator/ndt_scan_matcher/describe_parametersReply -> v1/rr/localization/pose_estimator/ndt_scan_matcher/describe_parametersReply): created with topic_type=rcl_interfaces::srv::dds_::DescribeParameters_Response_
    [zenoh-bridge-dds-70] [2023-07-25T07:52:08Z INFO  zenoh_plugin_dds] Route DDS->Zenoh (rr/localization/pose_estimator/ndt_scan_matcher/list_parametersReply -> v1/rr/localization/pose_estimator/ndt_scan_matcher/list_parametersReply): created with topic_type=rcl_interfaces::srv::dds_::ListParameters_Response_
    ```
    
    </details>

- For video2
    - [Additional video for video2](https://drive.google.com/file/d/11n3vhRK_n0rPt9Wjq0Y5HRrPYl-VgSGC/view?usp=sharing)
  <details>
[... cut ...]
  [zenoh-bridge-dds-70] [2023-07-25T07:42:32Z INFO  zenoh_plugin_dds] Route Zenoh->DDS (v1/rq/localization/pose_estimator/ndt_scan_matcher/list_parametersRequest -> rq/localization/pose_estimator/ndt_scan_matcher/list_parametersRequest): created with topic_type=rcl_interfaces::srv::dds_::ListParameters_Request_
  ```
  
  </details>

By the way, logs related to the bridge are also included, whose name is `zenoh-bridge-dds`.
And I don't know why but it seems like both logs are the same.
I just replayed each rosbag file and wrote the log regarding `ndt_scan_matcher`.
Should I rerun the scenario and log the message again?

kminoda, 2023-07-25:
OK, I got the point. At least from the video 1, the ndt_scan_matcher seems to work correctly, since the green points and the pointcloud map matches pretty well. Thus, I would say that it is an issue regarding other part, e.g. EKF. 
Indeed, it is a known issue with Autoware and still has not fixed yet (https://github.com/orgs/autowarefoundation/discussions/3554).

Kim-mins, 2023-07-25:
Thank you for clarifying the issue. I really appreciate to your help!

I checked the green and white points. Thank you.

By the way, I also observed that, from 1:02 of the video 1, I could see that the green arrow at the position of the ego vehicle **stops and does not follow the ego vehicle**.
I cannot explain why the green arrow should be aligned with the ego vehicle position, but it had been following the ego vehicle's position before the arrow stucks, so I thought that way. I want to ask if it is a normal behavior or not.

How do you think about video 2?
In my small opinion, each colored points are aligned well just like video 1. Would it also have the same root cause with that of video 1?

Kim-mins, 2023-07-25:
Oh then maybe you can refer to [this video](https://drive.google.com/file/d/1rTUGMVzsYk8BuPpfddPmpQscxVxECnXe/view).
From 0:12, two arrows has some disparity between them.
On 0:38, the green arrow does not follow up the ego vehicle's location.

I agree to your point. I also think that debugging this kind of complex software is not straightforward.. so I really appreciate to your help. It is really complicated job.

YamatoAndo, 2023-12-20:
@Kim-mins It has been a long time without any comments. Do you need assistance?
If it's already resolved, please close it.

Kim-mins, 2023-12-21:
Thank you for the remind @YamatoAndo!
I think it's OK now, so I'll close it. Thanks.

closed 2023-12-21 by Kim-mins