autowarefoundation/autoware_universe #4391: the path is not generated properly

beyzanurkaya, 2023-07-25:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The path is not appropriately generated in some cases. The generated path is being generated shorter than it should be.

https://github.com/autowarefoundation/autoware.universe/assets/32412808/711e9928-c56e-4175-aa8f-981db77cbc7a

### Expected behavior

This is how the path should look like:

[image]

### Actual behavior

The path looks like this:

[image]

### Steps to reproduce

1. Download [map.tar.gz](https://github.com/autowarefoundation/autoware.universe/files/12163055/map.tar.gz)
2. run planning simulator
3. give an init and goal pose

kosuke55, 2023-07-26:
@beyzanurkaya 
Thank you for your report. This issue seems similar to https://github.com/autowarefoundation/autoware.universe/pull/4109 :thinking: 
Does your branch include this PR? ( or similar issue happens again.

beyzanurkaya, 2023-07-26:
@kosuke55 similar issue but this problem doesn't occur only in u-turns

kosuke55, 2023-07-26:
memo:

here
[image]

NG
[image]

OK 
[image]

kosuke55, 2023-07-26:
seems similar but another problem to https://github.com/autowarefoundation/autoware.universe/pull/4109 :thinking:

kosuke55, 2023-07-26:
memo:

checked https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_path_planner/src/utils/start_planner/shift_pull_out.cpp#L150-L157

[image]

```
[component_container_mt-27] goal_is_behind: 0, s_start: 74.5449, s_end: 93.0561
[component_container_mt-27] road_lanes: 2039, 2062, 2071, 2100, 8350, 1663, 8401, 1747, 8462, 1801, 8580, 1816, 8457, 1786, 8405, 1674, 8364, 2270,
```

the road_lanes is wrong maybe because of getExtendedCurrentLanes.
need to check the extended lane is road_lanes and add that as possible.
https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_path_planner/src/utils/utils.cpp#L2857

kosuke55, 2023-07-27:
fixed in https://github.com/autowarefoundation/autoware.universe/pull/4418

[image]

beyzanurkaya, 2023-07-27:
@kosuke55 thank you, I'll close the issue then

mehmetdogru, 2023-07-27:
Lets close the issue after the PRs Kosuke-san opened. @beyzanurkaya please test the PRs if possible https://github.com/autowarefoundation/autoware.universe/pull/4418
https://github.com/autowarefoundation/autoware.universe/pull/4422

kosuke55, 2023-07-30:
I have merged the PRs, so please let me know if the problem still exists.

beyzanurkaya, 2023-07-31:
@mehmetdogru @kosuke55 I tested https://github.com/autowarefoundation/autoware.universe/pull/4418 and https://github.com/autowarefoundation/autoware.universe/pull/4422. This problem seems to no longer exist.

closed 2023-07-31 by mehmetdogru