autowarefoundation/autoware_universe #4425: object_merger removes obsacle

mebasoglu, 2023-07-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

With the default parameters, object_merger node removes the unknown obstacles around a known object. Even though `CAR-UNKNOWN` and `UNKNOWN-CAR` parameters in `can_assign_matrix` are activated, object_merger cannot merge them. Here are photos for the current situation. The first one shows the scenario. The obstacle inside the red circle is removed by object_merger.

[image]

Here is the RViz2 screen. The removed unknown object is in the red circle.

[image]

Here is the output of camera lidar fusion:

[image]

### Expected behavior

The unknown object in front of the ego vehicle should not be removed.

### Actual behavior

The unknown object in front of the vehicle is removed by the object_merger.

### Steps to reproduce

1. Launch centerpoint detection and camera_lidar_fusion.
2. Launch object_merger with the ouput of detections with default parameters.

### Versions

- OS: Ubuntu 22.04
- ROS 2: Humble

### Possible causes

object_merger cannot merge unknown objects and known objects properly. To remove unknown objects inside a known object, `remove_overlapped_unknown_objects` part was added. If the merging is good, there won't be need for `remove_overlapped_unknown_objects`.

miursh, 2023-08-04:
@mebasoglu 
#4499 is merged. Could you please check how this will look using the latest universe?

mebasoglu, 2023-08-04:
Hello, I tested it with a bag file and It worked, thank you. I am going to test it also on our vehicle and I will let you know as soon as I tried on the vehicle.

mebasoglu, 2023-08-07:
Hello, we have tested this and it works. We can close the issue. Here is an image of `object_merger` merges known and unknown objects successfully:

[image]