autowarefoundation/autoware_universe #4443: new ground_segmentation filter

yusufy1ld1z, 2023-07-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

There are many ground points in the output of the current ground filter. I have tried the algorithm accepted by RA-L with IROS'22 option named "TRAVEL: Traversable Ground and Above-Ground Object Segmentation using Graph Representation for 3D LiDAR Scans". The proposed algorithm is sensible and works better in accuracy and processing time than the scan_ground_filter. Due to the fact that this project is licensed with GPL-3.0, I am rewriting the code, making it more performant, and also adapting to Autoware stack by adding pointcloud_preprocessor, etc.
There is an comparison video of two algorithms : 
https://youtu.be/VkzRjY9xLdk

### Purpose

To give more precise output by the obstacle_segmentation/ground_filter with lowering the processing_time_ms.

### Possible approaches

Adding the travel_ground_filter among the other filters as an option

### Definition of done

- Rewriting the code to license it with Apache-2.0
- Adapt to Autoware stack and add some features like pointcloud_preprocessor
- Adjust the launch files by including the new method

ktro2828, 2023-07-28:
Looks grate! I'm actually implementing patchwork++ and I want to update it combining with TRAVEL. Therefore if you get any progress let's discuss about it!

https://github.com/ktro2828/autoware.universe/tree/feat/patchworkpp

yusufy1ld1z, 2023-07-28:
Thanks for your interest, excellent work!
I actually progressed quite much, I have written the implementation in the format of current ground filters, so I am working under the ground segmentation package rather than a distinct package because I am planning it to be an optional ground filter.

https://github.com/yusufy1ld1z/autoware.universe/tree/travel_ground_filter/perception/ground_segmentation

I will not be able to work on this project next week. I am waiting for your thoughts and future improvements with curiosity. After that, we can work together!

yusufy1ld1z, 2024-08-27:
Hello again, apologies for a long break. I have not changed anything since the last update because I thought I already finished the project when I opened this issue. I think @ktro2828 has done a well job but I could not find the package of him after the last updates in the Autoware. If you still interested in my job I am ready to make updates you desire. Initially I adopted my project as an option to ground filters already in the perception ground segmentation stack and I am still agree with that. What are your further thoughts about the project, I am looking forward to open a PR !

vividf, 2024-09-05:
@ktro2828 Do you have any idea about this?