autowarefoundation/autoware_universe #4482: ddsi_udp_conn_write to udp failed with retcode  -1 error shows up sometimes after launch

yaukaizhi, 2023-08-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Sometimes when launching the my sensor drivers with autoware, the following error pops up:

ddsi_udp_conn_write to udp failed <my ip address> with retcode  -1

### Expected behavior

no error messages

### Actual behavior

error messages as described

### Steps to reproduce

Launch autoware universe

maxime-clem, 2023-08-01:
Can you please share more details about the issue.
- Which driver is throwing this error ?
- Is there any way to reproduce the issue consistently ?
- How does the error affect Autoware ?

yaukaizhi, 2023-08-01:
- Error occurs when launching autoware universe after launching my own driver code
- Can't reproduce this consistently
- The error will cause autoware universe to crash

Just wanted to know if anyone has seen similar issues and what direction I should look into solving this :)

yaukaizhi, 2023-08-04:
solution is to write in .bashrc 
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp 

seems like cyclonedds cant handle large loads

yaukaizhi, 2023-11-01:
@maxime-clem I'm getting this issue when using autoware's recommended DDS settings. Has anyone encountered this?

**In the bashrc**
```
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
export CYCLONEDDS_URI=<path_to_cyclonedds_config.xml>
sudo sysctl -w net.core.rmem_max=2147483647
sudo sysctl net.ipv4.ipfrag_time=3
sudo sysctl net.ipv4.ipfrag_high_thresh=1342177280
```
**cyclone dds config xml**
```
<?xml version="1.0" encoding="UTF-8" ?>
<CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
<Domain Id="any">
        <General>
            <Interfaces>
                <NetworkInterface autodetermine="true" priority="default" multicast="default" />
            </Interfaces>
            <AllowMulticast>default</AllowMulticast>
            <MaxMessageSize>65500B</MaxMessageSize>
        </General>
        <Internal>
            <SocketReceiveBufferSize min="10MB"/>
            <Watermarks>
                <WhcHigh>500kB</WhcHigh>
            </Watermarks>
        </Internal>
    </Domain>
</CycloneDDS>
```
**Error messages I get**
```
[component_container_mt-57] 1698811787.925822 [0]        tev: ddsi_udp_conn_write to udp/192.168.1.119:53234 failed with retcode -1
[component_container_mt-57] 1698811787.925830 [0]        tev: ddsi_udp_conn_write to udp/192.168.1.119:53446 failed with retcode -1
[component_container_mt-57] 1698811787.925839 [0]        tev: ddsi_udp_conn_write to udp/192.168.1.119:53777 failed with retcode -1
[component_container_mt-57] 1698811787.925847 [0]        tev: ddsi_udp_conn_write to udp/192.168.1.119:54280 failed with retcode -1
[component_container_mt-57] 1698811787.925855 [0]        tev: ddsi_udp_conn_write to udp/192.168.1.119:55546 failed with retcode -1
```

maxime-clem, 2023-11-14:
I do not know about these error messages. Can you notice any issue when running Autoware which may be related to these messages ?
The error messages may be caused by one machine on your network (192.168.1.119). If you do not need to share message with that machine then you can ignore the error messages.

yaukaizhi, 2024-01-16:
The issue is caused by when your machine disconnects from the network it was on. I believe this could be a ROS2 thing.

closed 2024-01-16 by yaukaizhi

yaukaizhi, 2024-03-14:
我的问题是机器人断线后，ROS2 就没有IP了 就会看到这个

要确保不会断线