autowarefoundation/autoware_universe #4490: Autoware sometimes stuck in the middle of a parking scenario

TakaHoribe, 2023-08-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Even though the Planning module is still in the middle of the sequence, the goal decision is made with a large yaw angle error, and the state of Autoware is transited to STOP.

The essential cause is that the goal decision is made without considering the status of the Planning and Control.

In the following picture, the planning is still making the trajectory to fix the yaw angle, but the goal decision is made and the Autoware will not move.

Example 1
[image]

Example 2
[image]

### Expected behavior

The goal decision can appropriately consider the status of the planning process.

### Actual behavior

Goal decision is made in the middle of the planning process.

### Steps to reproduce

1. run Psim
2. test the freespace planner
3. the issue sometimes happen

### Versions

autoware.universe version: 2212b003ad5efb7f0baba221c976d0c8801ad236

TakaHoribe, 2023-08-01:
cc. @yukkysaito @isamu-takagi

isamu-takagi, 2023-08-03:
@TakaHoribe The mission_planner receives the goal pose and does arrival check with [its own parameters](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/mission_planner/config/mission_planner.param.yaml#L4-L6). Since this parameter does not reflect the control parameter, I think the arrival check should be moved to a path planning or control module that knows the control parameter.