autowarefoundation/autoware_universe #4498: control_performance_analysis crashes with segfault

kasper-helm, 2023-08-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I have seen `control_performance_analysis` crash when the `idx_prev_wp_` pointer is not initialized, and is dereferenced on https://github.com/autowarefoundation/autoware.universe/blob/main/control/control_performance_analysis/src/control_performance_analysis_core.cpp#L378

- this happens when the ego is either beyond `acceptable_max_distance_to_waypoint_` from the trajectory, or the heading error is larger than `acceptable_max_yaw_difference_rad_`, which causes [`closest_segment`](https://github.com/autowarefoundation/autoware.universe/blob/main/control/control_performance_analysis/src/control_performance_analysis_core.cpp#L92) to be empty, and the `idx_prev_wp_` to remain uninitialized 

### Expected behavior

The node should safely handle the case where `closest_segment` is empty 

### Actual behavior

[ERROR] [control_performance_analysis_exe-24]: process has died [pid 216, exit code -11, cmd ...

### Steps to reproduce

- I am using a pre-defined fixed `Trajectory` message to test the Autoware control component without the planning component
- This issue occurs when the localization initializes while the vehicle is somewhat far (> a few meters) from the trajectory

kyoichi-sugahara, 2024-02-17:
@kasper-helm 
Thank you so much for your report and sorry for late response :sob: 
If it's possible, could you give me the information how to reproduce problem?
(e.g. which version of autoware you are using, how did you launch autoware, which comman did you run to publish trajectory.

kasper-helm, 2024-02-20:
I think I fixed it by adding the following lines above https://github.com/autowarefoundation/autoware.universe/blob/main/control/control_performance_analysis/src/control_performance_analysis_core.cpp#L378

`  if (!idx_prev_wp_) {
    return std::nullopt;
  }`

kyoichi-sugahara, 2024-02-20:
@kasper-helm 
Wonderful thanks!
Sorry for not supporting well :cry: 
If you create PR, I will review the PR :+1:

theCodeForgerHQ, 2026-09-15:
Hi, I'd like to take this one. The crash path is still present on main: calculateErrorVars() ignores the result of findClosestPrevWayPointIdx_path_direction() (L145), so when ego starts off the trajectory idx_prev_wp_ is still null and findCurveRefIdx() dereferences it (L385). I'll open a PR with a small fix and the package's first unit test, which crashes on main and passes with the fix.